Audio-Injector/Octo #1: Random channel assignment and intermittent detection

Dloranger, 2018-03-27:
I am seeing intermittent detection of the octo card between boots, and random channel assignment within each iteration of speaker-test utility.

Note during these tests the hardware is sitting in isolation on my backplane board with the scope attached. Nothing is disturbing the hardware in any way except reboots which are done via command line and power cycles are done via killing the DC supply. I have checked the signals and everything is nice clean digital pulses coming from the pi3.

I am testing using the latest distro via apt-get update & apt-get upgrade (effective Mar 1, 2018).

I am using a customized version of the driver as discussed via email, where the clocks are never disabled. I don't expect this to be relevant, but just in case

The command I am running to test is 

`speaker-test -c 8 -t sine -f 500`

I am measuring what I consider channel 0 with an oscilloscope in the below tests

power cycle #1, card detected & loaded, signal detected during channel 2
reboot cycle #1, card detected, but not loaded into kernel
```
[ 9.551263] audioinjector-octo soc:sound: ASoC: CODEC DAI cs42448 not registered - will retry
[ 9.551278] audioinjector-octo soc:sound: snd_soc_register_card failed (-517)
reboot cycle #2, card detected, but not loaded into kernel
[ 7.941254] audioinjector-octo soc:sound: ASoC: CODEC DAI cs42448 not registered - will retry
[ 7.941270] audioinjector-octo soc:sound: snd_soc_register_card failed (-517)
```
reboot cycle #3, card detected & loaded, 
signal detected during channel 6 on speaker test session #1 
signal detected during channel 0 on speaker test session #2 
signal detected during channel 2 on speaker test session #3 
signal detected during channel 2 on speaker test session #4 
signal detected during channel 0 on speaker test session #5 
signal detected during channel 0 on speaker test session #6 // test killed while channel 5 was playing
signal detected during channel 4 on speaker test session #7 // test killed while channel 4 was playing
signal detected during channel 2 on speaker test session #8 // test killed while channel 2 was playing
signal detected during channel 6 on speaker test session #6 // test killed while channel 5 was playing

Below find the DTS and config.txt that I am using, I can't upload as files for some reason, invalid file format error

Here is the DTS that I am using which includes loading up 3x MCP23017 gpio expanders, these are all working perfectly.

```
/dts-v1/;
/plugin/;

/ {
compatible = "brcm,bcm2708";

fragment@0 {
target = <&i2s>;
__overlay__ {
status = "okay";
};
};

fragment@1 {
target = <&i2c1>;
__overlay__ {
#address-cells = <1>;
#size-cells = <0>;
status = "okay";

cs42448: cs42448@48 {
#sound-dai-cells = <0>;
compatible = "cirrus,cs42448";
reg = <0x48>;
clocks = <&cs42448_mclk>;
clock-names = "mclk";
VA-supply = <&vdd_5v0_reg>;
VD-supply = <&vdd_3v3_reg>;
VLS-supply = <&vdd_3v3_reg>;
VLC-supply = <&vdd_3v3_reg>;
status = "okay";
[... 326 lines cut ...]
002144.005: dtdebug: +prop(gpio-controller)
002145.662: dtdebug: +prop(#gpio-cells)
002147.337: dtdebug: +prop(interrupt-parent)
002148.972: dtdebug: +prop(interrupts)
002150.632: dtdebug: +prop(interrupt-controller)
002152.327: dtdebug: +prop(#interrupt-cells)
002154.042: dtdebug: +prop(microchip,irq-mirror)
002156.174: dtdebug: +prop(status)
002157.958: dtdebug: +prop(linux,phandle)
002160.102: dtdebug: +prop(phandle)
002161.856: dtdebug: merge_fragment() end
002161.898: dtdebug: merge_fragment() end
002161.991: dtdebug: Found fragment 7 (offset 1876)
002164.254: dtdebug: merge_fragment(/soc/gpio@7e200000,/fragment@7/__overlay__)
002164.282: dtdebug: +prop(pinctrl-names)
002166.607: dtdebug: +prop(pinctrl-0)
002173.129: dtdebug: merge_fragment(/soc/gpio@7e200000/my_pins,/fragment@7/__overlay__/my_pins)
002173.152: dtdebug: +prop(brcm,pins)
002175.332: dtdebug: +prop(brcm,function)
002177.528: dtdebug: +prop(brcm,pull)
002179.753: dtdebug: +prop(linux,phandle)
002182.351: dtdebug: +prop(phandle)
002184.546: dtdebug: merge_fragment() end
002184.575: dtdebug: merge_fragment() end
```

flatmax, 2018-03-27:
Hi there,

I think these problems are separate.

Intermittent detection is related to possible timing errors when 
resetting the codec, check this forum post out :

http://forum.audioinjector.net/viewtopic.php?f=9&t=3101

Where relevant code is here :

https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/audioinjector-octo-soundcard.c#L274-L287

A fix without rebooting and crossing your fingers is to run :
```
sudo modprobe -r snd_soc_audioinjector_octo_soundcard
sudo modprobe -r snd_soc_cs42xx8_i2c
sudo modprobe -r snd_soc_cs42xx8
sudo modprobe snd_soc_cs42xx8
sudo modprobe snd_soc_cs42xx8_i2c
sudo modprobe snd_soc_audioinjector_octo_soundcard
```
In your altered driver, you have removed clock turn on and turn off 
between play/capture sessions -
I have the feeling that if you don't shut down your clocks, you should 
keep the audio device running recording+playing from the first 
application till turn off. I can't see a way to stop random channel 
assignments otherwise.

Matt

Dloranger, 2018-03-27:
Okay, I can script the modprobe for now to get reliable detection, that will solve 1 problem.

Can you help to explain the mechanism that is causing the channels to be reassigned because the clocks are running?  I think this is fundamentally where the issue lies.  

On all other sound cards I have used, the left channel in hardware is always the left channel, and the right is always the right channel; they are never randomized.  This is somewhat out of my area of expertise, but feels like there is some configuration missing that would prevent the random channels. 

I can safely say the electrical interface in the chip is not changing, so it must be a software setting that is not doing the right things.  I don't know if this is in the audio codec chip drivers, alsa, or elsewhere though.

In the end, I need the clocks to be left running and channel X to always be channel X, regardless of reboots, power outages, application start and stop, etc.   I am more than happy to test whatever ideas you have if you can help me find a working configuration.  This is the last issue I am fighting with before being able to start launching our new design.

Appreciate the help so far, just stuck trying to get over this last hurdle

flatmax, 2018-03-27:
> Okay, I can script the modprobe for now to get reliable detection,
> that will solve 1 problem.
>
> Can you help to explain the mechanism that is causing the channels to
> be reassigned because the [...]

This card isn't stereo and it is running over a stereo bus. For that 
reason there we need to regain sync by stopping and starting the clocks.

Is there no way to change your software to leave capture/playback 
happening all the time ? This way you can start the clocks once and keep 
capture/playback happening till shutdown.

> I can safely say the electrical interface in the chip is not changing, 
> so it must be a software setting that is not doing the right things. I 
> don't know if this is in the audio codec chip drivers, alsa, or 
> elsewhere though.

This problem still hasn't been isolated and it isn't clear what the 
cause is. For now the rmmod and modprobe lines will get around this 
problem ... if you detect the card wasn't loaded run those commands and 
it will be detected.

Dloranger, 2018-03-27:
The software takes control of the rx and tx channels at launch, but the
play is on demand and can't be changed. The record is monitored
continuosly, but not sure why the clocks were not running automatically as
a result of this.

I asked the app developer about setting the audio device state and he
replied as follows..  perhaps you can see if there is something missing
that is needed to configure the card at launch that would resolve both
issues?

I tried but quickly got lost in the code.  It seems like you are looking
for some type of pcm state update or ?? That is not being triggered?

I don't recognize that there are any special code for ALSA power management
in SvxLink. All ALSA code is contained in this file
<https://github.com/sm0svx/svxlink/blob/master/src/async/audio/AsyncAudioDeviceAlsa.cpp>
if
you want to have a look at it.

https://github.com/sm0svx/svxlink/blob/master/src/async/audio/AsyncAudioDeviceAlsa.cpp

Dloranger, 2018-04-08:
I think I understand the issue with the sound synchronization now, but still not sure how to address the issue.

SVXLINK application is consuming the audio for the purposes of clocking.  I am not sure how this is happening, I can't follow the code enough to understand it, see above post for the link.  

My understanding from the behavior is the svxlink software takes control of the sound card somehow (as its blocked to other devices), but somehow that is not triggering your application to turn the clocking on except when svxlink software is doing playout functions.

I suspect the svxlink software is not doing a proper record, but rather monitoring the channels. The svxlink application is used to control communication repeaters that are expected to be left running for months to years without intervention (Some may be VERY remote from civilization) so a traditional record function wouldnt make sense..  Perhaps this is where the disconnect in the two usages does not overlap.  Just speculating here, but seems like a possibility.

Dloranger, 2018-04-08:
> If I recall correctly you wanted the clocks to be constantly on ? An
> attempt was made to hack the machine driver to make this happen, but
> then discovered this hack messed with the channel assignments.
>
> I Think I have a simpler solution for you - which starts by reverting
> back to the original machine driver.
>
> OK - is there a way to change the alsa device which the syvxlink
> software uses ?
>
> If so then you can have one application start playback/recording (of
> nothing i.e. zeros) on turn on. Then you can have the svxlink software
> use dsnoop and dmix alsa plugins as well to play/record. This way the
> card's clocks will always be on and there will be no channel assignment
> mixup.
>
>
>
> Matt

Yes, that is correct on the clocking, and I like the ideas, sounds a lot
cleaner.

I know I have the ability to select the card and channel in alsa for both
TX and RX.

I don't think that I can pick the pcm device directly to point to
dmix/snoop, but I might. I will have to investigate that more closely.

Let me research and see what I can do with this idea.

Dloranger, 2018-04-09:
So it appears that my octo card has died, it showed up on the i2c bus when
I first powered up the pi, but now cannot be detected after multiple
attempts, different pi3, fresh sd card, etc.

I hopped on to your store front to order another one, but looks like you
are out-of-stock.  Will you be getting more stock soon?  Also since I don't
need the audio adapter boards, just the codec board, any chance I can get
just that board from you?  In the mean-time I will replace the codec chip
in hopes of reviving this one.

Dloranger, 2018-04-09:
Okay, yikes on the few months but I completely understand. Lead times are
killing everyone lately, got a list of 125 parts with sourcing issues at
work

Replacing the codec did the job, glad I had a couple on hand.

Not sure on what would have caused it to fritz out, but it has been handled
a lot unprotected so esd is certainly possible. We have been showing the
prototype at shows. Lots of interest once we can get it all working.

I tried switching to the pcm of dsnooped and the application threw some
errors, not sure if it's a supported option or not, but appears to be
unsupported at the moment.

I posted to the developers support email group to ask the questions about
the configs and errors to see what they think about the findings.

> So it appears that my octo card has died, it showed up on the i2c bus when
> I first powered up the pi, but now cannot be detected after multiple
> attempts, different pi3, fresh sd card, etc.
>
> I hopped [...]

We are out of stock for a few months while we reset our FPGA process -
we are sourcing direct from the manufacturer this time which will make
manufacture more smooth in the future.

Replacing the codec may be a good start - any ideas on how it fried ?

https://github.com/

Dloranger, 2018-04-09:
I had an instance of where the sound card did not load, and I tried running the commands you provided above, and while the hardware is now claimed by the processor, alsa can't use it

pi@PI-REPEATER-2X:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: UU UU UU -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@PI-REPEATER-2X:~ $
pi@PI-REPEATER-2X:~ $
pi@PI-REPEATER-2X:~ $ sudo modprobe -r snd_soc_audioinjector_octo_soundcard
pi@PI-REPEATER-2X:~ $ sudo modprobe -r snd_soc_cs42xx8_i2c
pi@PI-REPEATER-2X:~ $ sudo modprobe -r snd_soc_cs42xx8
pi@PI-REPEATER-2X:~ $ sudo modprobe snd_soc_cs42xx8
pi@PI-REPEATER-2X:~ $ sudo modprobe snd_soc_cs42xx8_i2c
pi@PI-REPEATER-2X:~ $ sudo modprobe snd_soc_audioinjector_octo_soundcard
pi@PI-REPEATER-2X:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: UU UU UU -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@PI-REPEATER-2X:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
equal
anyChannelCount
default
pi@PI-REPEATER-2X:~ $ aplay -l
aplay: device_list:270: no soundcards found...
pi@PI-REPEATER-2X:~ $ alsamixer
cannot open mixer: No such file or directory
pi@PI-REPEATER-2X:~ $

This is what I did just before it disappeared from the system last time in case it matters

nettings, 2019-01-17:
@Dloranger, sorry to bump this old thread, but have you considered using JACK for your application? That way, the device can keep running all the time (with clocks) and retain its channel order reliably. It's what I'm doing here with a number of different I²S soundcards, and it works very well.

Dloranger, 2019-01-17:
@nettings,

I havent used Jack as I dont have a gui.  The application runs headless, but if you could advise a settings set to get started with?

Perhaps I could set up a station with a gui to get the config file and then deploy it to the headless setups?

This is well outside my expertise so any guidance is much appreciated.

nettings, 2019-01-17:
JACK does not require a GUI. I'm using it headless here, configured so that the system comes up with all audio stuff running without user interaction. 

```
my_user@mn_29f4ff:~ $ cat /etc/systemd/system/jackd.service 
[Unit]
Description=JACK Audio Connection Kit
After=sound.target
After=ntp.service
After=time-sync.target
Before=jackd.target
Requires=jackd.target

[Service]
EnvironmentFile=-/etc/my_jack_config
LimitRTPRIO=85
LimitMEMLOCK=700000000
User=my_user
Environment="DBUS_SESSION_BUS_ADDRESS=unix:path=/run/dbus/system_bus_socket"
ExecStartPre=/bin/sleep 10
ExecStart=/usr/bin/jackd $JACKD_OPTIONS
RestartSec=5
Restart=on-failure
```

In this case, $JACKD_OPTIONS is parsed from an external config file my_config. If your system hardware is fixed, you might as well hardcode it.
For the Octo, I'm using "-R -P70 -dalsa -dhw:audioinjectoroc -r48000 -p256 -n3 -zs".
Note the DBUS variable which is necessary if you want to run as a non-root user without a login session. For testing, don't use this right away, try cmdline first. When it runs, `jack_lsp` will show you the available ports, `jack_connect` connects them. Note that if you're currently pushing audio to the interface, you will have to re-design your application to implement a callback, because jack will tell you when to send audio.

As for the systemd integration, check http://jack-audio.10948.n7.nabble.com/Jack-Devel-Some-observations-re-Jack-and-systemd-td20340.html.