Audio-Injector/Octo #2: Octo with banana pi m2 berry/ultra

K3nn3th2, 2018-03-27:
Hey folks, i wanted to add this to the thread i had already started last year regarding the banana pi m3.
I saw the forums had moved so i thought it was appropriate to start a new issue here.

According to the docs (https://bananapi.gitbooks.io/bpi-m2-ultra-open-source-single-board-computer/content/bpi-m2ultragpiopindefine.html) the banana pi m2 berry has the same gpio layout as the RPI b+.
Does this facilitate compatibility with the octo?

We still need the alsa machine drivers right?

flatmax, 2018-03-27:
There is potential to make it work.

The relevant drivers which need to be compiled with the Banana Pi m2 
berry/ultra's kernel are the octo machine driver :

https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/audioinjector-octo-soundcard.c

Also the device tree overlay :

https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/audioinjector-addons-overlay.dts

In the dts file, make sure the gpio lines match those (in software) on 
the Banana pi.

In the machine driver, these lines :

https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/audioinjector-octo-soundcard.c#L262-L267

Specify which nodes are sourced from the i2s device tree (called 
"i2s-controller") and the "codec" is named in the 
/audioinjector-addons-overlay.dts. Here are the lines which call for 
device tree nodes :

https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/audioinjector-octo-soundcard.c#L262-L267

Once you have set these parts up and they are working, then we have to 
hope that the syncrony is set correctly for the banana pi to work !

Matt

K3nn3th2, 2018-04-09:
hm.. i can only find older kernels for the bpim2u. 
i'll try compiling my own.

is there a possibility to add support for the octo in older kernel releases, like a patch?

K3nn3th2, 2021-07-22:
is the octo driver not included in the mainline kernel by now..?
would this make things easier?

flatmax, 2021-07-22:
it is included

K3nn3th2, 2021-08-11:
i am running armbian with a 5.11 kernel on my bpi m2u just now and trying to get the octo running with it.
according to the bpi m2u GPIO layout, it is the same as the  rpi's.
i got the octo .dts file, updated the compatibility string to include "allwinner,sun8i-r40", compiled it and am loading the overlay, [as described here.](https://docs.armbian.com/User-Guide_Allwinner_overlays/)

the octo is not recognized yet, though...

do you have any hints what i could do?