Audio-Injector/Octo #5: Octo sound card directly to fpga

gdragoi14, 2018-04-07:
Hi,

I want to connect the octo soundcard directly to an fpga board like a spartan7 via gpio/pmod (or other is there a better way?). I can obviously do this through an rpi connected to the fpga but i would prefer to do my audio processing without the extra hop. Is there any reason why this wouldn’t work?

Cheers,
George

flatmax, 2018-04-08:
> Hi,
>
> I want to connect the octo soundcard directly to an fpga board like a
> spartan7 via gpio/pmod (or other is there a better way?). I can
> obviously do this through an rpi connected to the fpga [...]

I think it can work. You will need to make sure you have the GPIO from 
here setup correctly :
http://forum.audioinjector.net/viewtopic.php?f=9&t=1317
You will have to reference the machine driver setup for difference 
sampling rates.
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/sound/soc/bcm/audioinjector-octo-soundcard.c

I guess you will use i2c to control the codec ?

gdragoi14, 2018-04-08:
Awesome, thank you so much!! Can't wait to pick one or two of these up when they become available again

Cheers,
George