Audio-Injector/Octo #6: Octo audio injector drawing?

jefflapierre6, 2018-04-09:
Where can I find a mechanical drawing of this board?

flatmax, 2018-04-09:
On 10/04/18 02:38, Jeff Lapierre wrote:
>
> Where can I find a mechanical drawing of this board?
>
>

I have forked the kicad PCB here :
https://github.com/Audio-Injector/AudioInjector.Octo.template

It shows the mechanical layout for a breakout board setup. The PCB edges 
match the card's edges, including the holes.

jefflapierre6, 2018-04-10:
I have a v1.16 board, which looks different. Most important is that it does not include the Raspberry Pi pass through pins. I am designing a board that fits onto both headers and need to have precise measurements. Thank you!

flatmax, 2018-04-11:
When you say "pass through pins, are you talking about the GPIO header pins ?

jefflapierre6, 2018-04-11:
Yes that is what I meant.

jefflapierre6, 2018-04-19:
Another unrelated question. I am developing a sensor that uses your audio injector as a digitizer. I had not predicted that this board would go out of stock and am in need of more to move forward with this project. Is there any way we can get more? Could you make a special order somehow? If not, would you be willing to provide the details of the board so that we can manufacture them independently? I could sign some sort of nondisclosure agreement if you would like. We could continue this conversation through email.