Audio-Injector/Octo #7: Let me know when the Octo is back in stock

TechnoHomesteader, 2018-04-16:
Placing an issue here as a pseudo stock-alert for the Octo. Please mark this resolved/closed when the Octo comes back into stock.

Thanks!

flatmax, 2018-04-16:
Sure - will do.
Background on this issue...
We are changing out FPGA sourcing process and this has delayed production by a couple of months. This will solve a lot of problems for future supply, and for that reason the disruption was necessary. We don'y expect this type of disruption to supply in the future again.

gfilippi, 2018-04-18:
Hi,
I'm adding myself to the list. I will check on this issue to be closed when  the Octo is back in stock. Kudos to Max for this great device. Great idea, smart execution.

omarmarcillo, 2018-05-01:
Adding myself to the list too. I am using the stereo sound card and it is great!. I am looking forward to trying the Octo.

Dloranger, 2018-05-18:
Add me to the list as well

flatmax, 2018-05-26:
Just a little update - we are expecting to receive new FPGA stocks late June, so it it likely we will have these selling again late July.

zephyrr, 2018-06-21:
Also interested

JohnAbarro, 2018-06-22:
Yess! Add me to the list as well, please :D

jefflapierre6, 2018-06-27:
Adding myself to this since I am interested in buy a bunch of these (~50).

fred-dev, 2018-07-08:
Yes, I want one too, let us know when they are available.

gfilippi, 2018-07-16:
Hi,
sorry to ask again .. do we have an ETA on the new FPGA batch (and so stock availability) ?
apr->july we should be almost there ... :-)

gfilippi, 2018-07-23:
Hi all,
I really need to buy few octo .. could you please post an update on the status of the FPGA? Are we expecting to have the card back in stock around AUG or is it going to be the end of 2018? thanks

flatmax, 2018-07-23:
We expect some Octo on stock today.

Please check the shop:

http://shop.audioinjector.net