Audio-Injector/Octo #10: octocard setup problem using deb

omarmarcillo, 2018-08-03:
Hi,

I just got an octocard but I get this error:

sudo dpkg -i audioinjector.octo.setup_0.3_all.deb 
dpkg-deb: error: 'audioinjector.octo.setup_0.3_all.deb' is not a debian format archive
dpkg: error processing archive audioinjector.octo.setup_0.3_all.deb (--install):
 subprocess dpkg-deb --control returned error exit status 2
Errors were encountered while processing:
 audioinjector.octo.setup_0.3_all.deb

I got the deb file from github, I also dowloaded the file from the audioinjector webpage and I get same message. Is there something I am missing?

REgards

Omar

jefflapierre6, 2018-08-07:
I just tried to install this package and got the exact same message.

jefflapierre6, 2018-08-07:
Figured it out. You can't just right-click and download the link. You need to click on it, which brings you to the file page, and then click on the 'download' button. It installed fine after that.