Audio-Injector/Octo #13: Octo detected, but no sound output

JohnAbarro, 2018-09-02:
The octo seems to be detected by my raspberry pi 3b. However, I'm not getting any output signals on any channels. I've followed each and every instruction pertaining to setting it up. It's showing up on dmesg as you have shown in the video. I've checked my asoundrc file and it's the same as the setup. I have removed pulseaudio and the volume plugin for alsamixer. I'm still not getting anything out. I've verified that the file config follows the anychannelcount plugin, which based on my analysis of the code, should allow to play on any of the eight channels of octo. I tried rebooting and I have been up testing this for two days. My doubts are leaning on the possibility of either the soundcard or the breakout being broken. Barely even used and never got any signs that the soundcard was making an output at all, other than the fpga chip getting hot when its processing sound.

I was thinking about using the oscilloscope to test the output terminals on the board where the ribbon is connected, but I need a diagram to know where the pins should be. Would you kindly give me a diagram for the pins on the octo board?

But since I'm running out of time, I am most likely to want to replace the sound card. Would it be possible to send the octo back for a replacement?

flatmax, 2018-09-02:
I understand.

Please contact matt@audioinjector.net

closed 2018-09-02 by flatmax

flatmax, 2018-11-06:
We have been talking on email - but for others, here is the RCA breakout 
repo :

https://github.com/flatmax/AudioInjector.Octo.RCA