Audio-Injector/Octo #14: Octo not working anymore with RPi kernel 4.19

Mohawi, 2019-01-04:
Hi Matt, 

I recently purchased an octo card and wanted to use it with my RPi 3 running libreELEC v8.95.y

When trying I noticed, that the device is not working correctly (is not detected) in this setup. The codec seems not registered. You can find the kernel ring buffer dump here: http://ix.io/1xqH

The card works like a charm with an older libreELEC version (8.2.5) though.

Trying the card with raspian running the Kernel 4.19 produces the same error messages.

According to "Hasian" from libreELEC forum, this problem seems to be associated with the octo driver not compiling correctly under RPi Kernel 4.19. So they removed it temporarely.
-> see https://forum.libreelec.tv/thread/14074-setup-audio-card-results-in-codec-not-registered-message/

Can you please look into this problem and hopefully commit an adopted version of your driver?

Best regards and thank you,
mohawi

flatmax, 2019-01-29:
I have looked into this, I believe this pull request against the Pi Kernel should fix the problem. You should be able to bring the card up by soft restarting (no power cycle).

Here is the commit/pull request we are waiting on :
https://github.com/raspberrypi/linux/pull/2834

pelwell, 2019-01-29:
Now merged.

flatmax, 2019-01-31:
Thank you for the support Phil.

Mohawi, 2019-02-02:
Hi, unfortunately I'm still not able to get the card working with the 4.19 Kernel. Maybe I do something wrong?
-> see discussion on libreelec forum

Mohawi, 2019-02-10:
Any news yet?

Mohawi, 2019-03-14:
I'm sorry to ask again, but have there been any new developments or new insights to the problem?

flatmax, 2019-03-14:
Now that the mainline kernel has updated to 4.19 I will address the problem with the 4.19 kernel and the octo.

For now you can downgrade the kernel. For example :

sudo rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b

Will take you to the 4.14.98 kernel version (from one month ago) :

https://github.com/Hexxeh/rpi-firmware/commit/a08ece3d48c3c40bf1b501772af9933249c11c5b

flatmax, 2019-03-20:
Initial fix for this bug is to load the snd_soc_cs42xx8_i2c module - which isn't auto-loading for some reason.

The best way is to load the module using the modules file. Edit /etc/modules and add the following line :
`snd_soc_cs42xx8_i2c
`
This makes your modules file look like the following :
pi@raspberrypi:~ $ cat /etc/modules
```
i2c-dev
snd_soc_cs42xx8_i2c
```

Another way to fix this is with the following script :

```
#!/bin/bash
# script to restart the octo 

sudo modprobe -r snd_soc_audioinjector_octo_soundcard
sudo modprobe -r snd_soc_cs42xx8_i2c
sudo modprobe -r snd_soc_cs42xx8
sudo modprobe snd_soc_cs42xx8
sudo modprobe snd_soc_cs42xx8_i2c
sudo modprobe snd_soc_audioinjector_octo_soundcard

```

pelwell, 2019-03-21:
Hi Matt,

There are a few things going wrong here, and although I've got it working, I'm not completely happy.

There is a significant change in 4.19 that creates DT aliases when an I2C alias is registered - see https://github.com/raspberrypi/linux/commit/af503716ac1444db61d80cb6d17cfe62929c21df. The problem with the cs42xx8 driver is that the base driver registers the OF/DT aliases while the cs42xx8-i2c shim registers I2C aliases. Now that the I2C subsystem is also generating OF uevents, the base driver gets loaded, but because it doesn't export the correct entry points (they're in the -i2c shim) it gets ignored. My fix is to move the declaration of the DT aliases to the -i2c shim, in the case where it is enabled.

There's also a bug in one of the error paths of your driver, which checks `dai->cpu_of_node` when it will never have been set - `i2s_node` is the value you want. I also see warnings about the overlay because the clock is defined within the I2C controller node, even though it isn't an I2C device. While I was fixing that I also sorted the nodes in an attempt to get runtime dtoverlay application to work, but it still doesn't (for some reason the phandle is missing - a problem for another day).

See https://github.com/raspberrypi/linux/pull/2902 for my proposed solutions.

flatmax, 2019-03-21:
> I2C alias is registered - see raspberrypi/linux@af50371
> See raspberrypi/linux#2902

Thanks for the patch - that is a great work around. I'll test it ASAP.

Matt

Mohawi, 2019-03-22:
@Matt many thanks for the quick workaround and for the coming fix 🙂

Mohawi, 2019-04-23:
Great, many thanks to @pelwell

closed 2019-04-23 by Mohawi

nettings, 2019-04-23:
@Mohawi, I have built a kernel with pelwell's fixes for Pi2 and newer, it seems to work reliably. 
I can send you a tarball if you want.

Mohawi, 2019-04-23:
@nettings, many thanks. That would be amazing.

nettings, 2019-04-23:
https://luchtbeweging.nl/download/rpi-4.19.34-v7+pelwell_octofix.tar.bz2
It unpacks to a folder "kernel", that's your root. It contains subfolders "boot/" and "lib/", the contents of which must go into your /boot and /lib. I'm assuming you have recently updated your firmware, if not, please run "rpi-update" before installing it. Also, it's v7+ only, i.e. it won't run on a Pi 1.

nettings, 2019-04-23:
Or, as per pelwell's estimate from https://github.com/raspberrypi/linux/pull/2834#issuecomment-485745336 , you can just wait until the end of the week and pull it with a regular rpi-update.

nettings, 2019-04-23:
In fact, it already got included, so you can ```rpi-update``` right now, 4.19.36-v7+ contains the fix.