Audio-Injector/Octo #15: Error on install

DareDrop, 2019-01-21:
Error on install. RasPi 3 B, OS 2018-11-13-raspbian-stretch-full

pi@raspberrypi:~ $ sudo dpkg -i Downloads/Octo-master/audioinjector.octo.setup_0.3_all.deb
Selecting previously unselected package audioinjector.octo.setup.
(Reading database ... 133377 files and directories currently installed.)
Preparing to unpack .../audioinjector.octo.setup_0.3_all.deb ...
Unpacking audioinjector.octo.setup (0.3) ...
Setting up audioinjector.octo.setup (0.3) ...
updating the kernel
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 13545  100 13545    0     0  49334      0 --:--:-- --:--:-- --:--:-- 49434
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 4.14.79-v7+
#############################################################
This update bumps to rpi-4.14.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    478      0 --:--:-- --:--:-- --:--:--   480
100 53.7M  100 53.7M    0     0  2093k      0  0:00:26  0:00:26 --:--:-- 3076k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.14.93+
 *** depmod 4.14.93-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 31e5fbf1fb44b2be60499b3bc452fd8a7fd2281f
 *** A reboot is needed to activate the new firmware
setting up /boot/config.txt
backing up /boot/config.txt to /boot/config.txt.2019-01-21.15.25.07
setting up /boot/config.txt
backing up /etc/asound.conf to /etc/asound.conf.2019-01-21.15.25.07
backing up /root/.asoundrc to /root/.asoundrc.2019-01-21.15.25.07
backing up /home/pi/.asoundrc to /home/pi/.asoundrc.2019-01-21.15.25.07
lxpanel's volume plugin is disfunctional, disabling it
sed: can't read /home/pi/.config/lxpanel/LXDE-pi/panels/panel: No such file or directory
dpkg: error processing package audioinjector.octo.setup (--install):
 subprocess installed post-installation script returned error exit status 2
Errors were encountered while processing:
 audioinjector.octo.setup
pi@raspberrypi:~ $

flatmax, 2019-01-28:
Hi there, I am working through this problem.

The auto install script breaks at the very end as you point out. This is ok as everything required is already setup. However there is a problem in the latest kernel.

It will work as expected if you wind back the kernel to early this month (kernel version 4.14.92) :
sudo rpi-update 6aec73ed5547e09bea3e20aa2803343872c254b6

stay tuned ...

flatmax, 2019-01-29:
Can you please do a test when it fails ? Can you please type sudo reboot or soft reboot the system (without cycling the power) ... does the board show up ?

pelwell, 2019-01-29:
Now merged.

flatmax, 2019-01-31:
Thank you,

May your beverages give you much joy.