Audio-Injector/Octo #17: doubt about using 2 octo to increase to 12 in and 16 out

alvarxo, 2019-03-04:
hello, my question is if I can use 2 octo cards for the same raspberry, or in their defender, use a second raspberry as a slave, but that the audio programs of the raspberry master recognize the 12 in and 16 out

flatmax, 2019-03-04:
You can only do this with 2 Pi boards.

Each Pi can only connect to one I2S bus. For that reason, for more the 
6/8 channels, you will require two Pi boards.

Matt

alvarxo, 2019-03-04:
ok, but is there any way to use them in such a way that it works as a single raspberry with 12/16?

flatmax, 2019-03-05:
Not that I am aware of.

nettings, 2019-04-23:
@alvarxo, you might want to check out JACK and zita-njbridge. Run one JACK on each RPi with an Octo, and zita-njbridge allows them to exchange audio over ethernet. Getting them time-aligned across both Pis is a bit of work, though, and you will have to live with higher latency.