Audio-Injector/Octo #18: octo detected, but no sound

mkafaudio, 2019-05-29:
dear flatmax,

when trying to install JACK with octo-soundcard i failed to get it working with the default setup of audioinjector-website and inputs from the forum. even speaker-test -c8 is not working and hangs.
pls see information attached.
is there anything missing?

best regards
markus

>>>>>>>>>>>>>>>Qjackctl status window
00:40:53.460 Statistics reset.
00:40:53.471 ALSA connection change.
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
inotify_add_watch("/home/pi/.config/qt5ct") failed: "No such file or directory"
00:40:53.581 ALSA connection graph change.
00:40:53.685 ALSA active patchbay scan...
00:41:17.988 JACK is starting...
00:41:17.991 /usr/bin/jackd -dalsa -dhw:audioinjectoroc,0 -r48000 -p1024 -n3 -P -o8
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
00:41:18.083 JACK was started with PID=1217.
jackdmp 1.9.11
Copyright 2001-2005 Paul Davis and others.
Copyright 2004-2014 Grame.
jackdmp comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions; see the file COPYING for details
JACK server starting in realtime mode with priority 10
self-connect-mode is "Don't restrict self connect requests"
audio_reservation_init
Acquire audio card Audio0
creating alsa driver ... hw:audioinjectoroc,0|-|1024|3|48000|0|8|nomon|swmeter|-|32bit
configuring for 48000Hz, period = 1024 frames (21.3 ms), buffer = 3 periods
ALSA: final selected sample format for playback: 32bit integer little-endian
ALSA: use 3 periods for playback
00:41:25.156 Could not connect to JACK server as client. - Overall operation failed. - Server communication error. Please check the messages window for more info.
JackPosixProcessSync::LockedTimedWait error usec = 5000000 err = Connection timed out
Driver is not running
Cannot read socket fd = 17 err = Success
CheckRes error
JackSocketClientChannel read fail
Cannot create new client
Cannot open qjackctl client
JackShmReadWritePtr1::~JackShmReadWritePtr1 - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
00:41:27.315 JACK is stopping...
Jack main caught signal 15

>>>>>>>>>>>>>>>>><>alsa-info.sh report
cat alsa-info.txt 
upload=true&script=true&cardinfo=
!!################################
!!ALSA Information Script v 0.4.64
!!################################

!!Script ran on: Tue May 28 16:58:52 UTC 2019
!!Linux Distribution
!!------------------
ID_LIKE=debian
!!DMI Information
!!---------------
Manufacturer: 
[... 649 lines cut ...]
Stream parameters are 48000Hz, S16_LE, 8 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 32 to 32768
Period size range from 16 to 16384
Using max buffer size 32768
Periods = 4
was set period_size = 8192
was set buffer_size = 32768
0 - Front Left

^CWrite error: -4,Interrupted system call
xrun_recovery failed: -4,Interrupted system call
Transfer failed: Interrupted system call

**>>>>>>and without sudo
speaker-test -c8**

speaker-test 1.1.3

Playback device is default
Stream parameters are 48000Hz, S16_LE, 8 channels
Using 16 octaves of pink noise
ALSA lib pcm_hw.c:1789:(_snd_pcm_hw_open) Unknown field slave
Playback open error: -22,Invalid argument

flatmax, 2019-05-29:
> [ 5.253918] audioinjector-octo soc🔉 cs42448 <-> 3f203000.i2s mapping ok

Hi there,

Can you please show the output of :

aplay -l

cat /proc/version

mkafaudio, 2019-05-29:
pi@Qtrack_octo:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: audioinjectoroc [audioinjector-octo-soundcard], device 0: AudioInject-HIFI cs42448-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@Qtrack_octo:~ $ cat /proc/version
Linux version 4.19.46-v7+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1230 SMP Tue May 28 16:18:48 BST 2019

best regards
markus

flatmax, 2019-07-17:
ok - lets backtrack a little.

The device is recognised - that is good.

Have you confirmed that you .asoundrc matches that from the setup guide ?

What happens when you play a regular 2 channel audio file with aplay ?

mkafaudio, 2019-07-23:
Dear Max,

thank you very much for your follow up!

i am on vacation and will come back with the requested information till end of the week.

best regards
markus

mkafaudio, 2019-08-01:
Dear Max,

the situation did not change in the meantime.

i am using a fresh install of buster Linux Jocto 4.19.60-v7+ #1247 SMP Thu Jul 25 14:41:19 BST 2019 armv7l GNU/Linux

and the  audioinjector.octo.setup_0.4_all.deb  and install script 

all i found is that there are no snd modules loaded

it looks like playing, but still no sound

best regards
markus

flatmax, 2019-08-01:
strange. It works with standard raspbian - right ?

aplay -l shows the same thing ?
alsamixer works ?

mkafaudio, 2019-08-01:
good evening Max,

yes, i can’t believe it also. 

1. aplay shows

pi@Jocto:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: audioinjectoroc [audioinjector-octo-soundcard], device 0: AudioInject-HIFI cs42448-0 []
  Subdevices: 0/1
  Subdevice #0: subdevice #0

pi@Jocto:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
anyChannelCount
default
sysdefault:CARD=audioinjectoroc
    audioinjector-octo-soundcard, 
    Default Audio Device
dmix:CARD=audioinjectoroc,DEV=0
    audioinjector-octo-soundcard, 
    Direct sample mixing device
dsnoop:CARD=audioinjectoroc,DEV=0
    audioinjector-octo-soundcard, 
    Direct sample snooping device
hw:CARD=audioinjectoroc,DEV=0
    audioinjector-octo-soundcard, 
    Direct hardware device without any conversions
plughw:CARD=audioinjectoroc,DEV=0
    audioinjector-octo-soundcard, 
    Hardware device with all software conversions

pi@Jocto:~ $ amixer
Simple mixer control 'ADC High-Pass Filter',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'ADC Soft Ramp & Zero Cross Control',0
  Capabilities: enum
  Items: 'Immediate Change' 'Zero Cross' 'Soft Ramp' 'Soft Ramp on Zero Cross'
  Item0: 'Immediate Change'
Simple mixer control 'ADC1',0
  Capabilities: cvolume
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 176
  Front Left: Capture 128 [73%] [0.00dB]
  Front Right: Capture 128 [73%] [0.00dB]
Simple mixer control 'ADC1 Invert',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [off]
  Front Right: Playback [off]
Simple mixer control 'ADC1 Single Ended Mode',0
  Capabilities: enum
  Items: 'Differential' 'Single-Ended'
  Item0: 'Differential'
Simple mixer control 'ADC2',0
  Capabilities: cvolume
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 176
  Front Left: Capture 128 [73%] [0.00dB]
  Front Right: Capture 128 [73%] [0.00dB]
Simple mixer control 'ADC2 Invert',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [off]
  Front Right: Playback [off]
Simple mixer control 'ADC2 Single Ended Mode',0
  Capabilities: enum
  Items: 'Differential' 'Single-Ended'
  Item0: 'Single-Ended'
Simple mixer control 'ADC3',0
  Capabilities: cvolume
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 176
  Front Left: Capture 128 [73%] [0.00dB]
  Front Right: Capture 128 [73%] [0.00dB]
Simple mixer control 'ADC3 Invert',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [off]
  Front Right: Playback [off]
Simple mixer control 'ADC3 Single Ended Mode',0
  Capabilities: enum
  Items: 'Differential' 'Single-Ended'
  Item0: 'Single-Ended'
Simple mixer control 'DAC Auto Mute',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'DAC De-emphasis',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]
Simple mixer control 'DAC Soft Ramp & Zero Cross Control',0
  Capabilities: enum
  Items: 'Immediate Change' 'Zero Cross' 'Soft Ramp' 'Soft Ramp on Zero Cross'
  Item0: 'Immediate Change'
Simple mixer control 'DAC1',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 255 [100%] [0.00dB]
  Front Right: Playback 255 [100%] [0.00dB]
Simple mixer control 'DAC1 Invert',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [off]
  Front Right: Playback [off]
Simple mixer control 'DAC2',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 255 [100%] [0.00dB]
  Front Right: Playback 255 [100%] [0.00dB]
Simple mixer control 'DAC2 Invert',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [off]
  Front Right: Playback [off]
Simple mixer control 'DAC3',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 255 [100%] [0.00dB]
  Front Right: Playback 255 [100%] [0.00dB]
Simple mixer control 'DAC3 Invert',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [off]
  Front Right: Playback [off]
Simple mixer control 'DAC4',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 255 [100%] [0.00dB]
  Front Right: Playback 255 [100%] [0.00dB]
Simple mixer control 'DAC4 Invert',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [off]
  Front Right: Playback [off]
Simple mixer control 'Mute ADC Serial Port',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [off]

2. alsamixer shows

best regards
markus

flatmax, 2019-08-06:
It looks like the DAC gains are all low ?
The alsamixer image didn't come through but that is alright.

BruceSpringsteen, 2020-01-10:
Hi.  Did this issue get solved?  Im having the same problem.  

Can see the device:

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: audioinjectoroc [audioinjector-octo-soundcard], device 0: AudioInject-HIFI cs42448-0 [AudioInject-HIFI cs42448-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

Alsamixer and soundcard UI gains are all up but no sound from any outputs.  Im not much of a coder but if there are any commands to shed some more light I'd appreciate the help.