Audio-Injector/Octo #19: Octo Soundcard supplies phantom power?

ishaaniwani, 2019-08-06:
Hello, I have 3 xlr microphones, that I plan to use a cable to turn into L and R RCA inputs. In order to record, the microphone needs power from the XLR cable. That means the sound card will have to send the power through the RCA inputs to the XLR microphones. 

Does the Octo Soundcard support the capability? If so, how much power does it give?

flatmax, 2019-08-06:
The Octo doesn't supply voltage over the codec lines.

Is your microphone an electret or condenser microphone ?

Electret microphones typically require less power then condenser 
microphones.

If you have condenser microphones then you will want to source a phantom 
power preamplifier.

If you have electret microphones, you will want to use an electret 
microphone preamplifier.

As microphones have low signal levels, you will also want to make sure 
they have enough preamplifier gain (or get a second preamplifier unit). 
Typically you can use a preamp with around 30 dB gain.

nettings, 2020-03-23:
@ishaaniwani, even if the Octo had this feature, it is impossible to have XLR phantom power from an RCA output. Phantom power requires a balanced (i.e. three-wire) connection.

flatmax, 2020-03-23:
The isolated sound card introduces phantom power and a standardised analogue skin. The skin includes phantom power, you can see Vph here :
https://github.com/Audio-Injector/AudioInjector-Analogue-Breakout-Standard/blob/master/AudioInjectorAnaBreakStdv1.pdf

nettings, 2020-03-24:
Good to know. The OP however asked about the "RCA inputs", so I was assuming they used the default RCA breakouts. Are you still offering the balanced breakouts pre-made, or is it DIY only?