Audio-Injector/Octo #20: Octo for Raspberry Pi 4?

jefflapierre6, 2019-08-15:
Sorry if this is not the right place for this, but I was curious if there are plans for you to make an upgraded version of the octo board for the Raspberry Pi 4?

flatmax, 2019-08-18:
When I last looked, the device tree seemed to indicate that the Pi4 has 
the same I2S silicon as the Pi 3 ?

Have you tried an Octo with the Pi4 ?

Matt

jefflapierre6, 2019-08-19:
I haven't checked that (I don't have a Pi4), but I read that the shape of the Pi4 did not match that of the Pi3, so I assumed it wasn't the same. Probably not a good assumption.

BennPaul, 2019-08-19:
I just tried the octo on a pi 4 with raspian buster and with the automatic setup. I tested the output channels with mixxx.org and all output channels worked. 

But there was a really strange behaviour when trying to also set the input channels: The output channels get all mixed up and some stopped working. I didn't had the time to have a closer look. I'll update you asap

flatmax, 2019-08-22:
I think the silicon is the same for I2S between the Pi 3 and Pi4 last time I checked ... it is likely that the Octo will work as expected on both platforms.

keygee, 2019-10-09:
Hi all,
I'm successfully using my octo on a PI4.
It is working great on my custom linux (crux-arm) from PI2 => PI3 => PI4 using kernel 4.19.75
Thanks Flatmax and all kernel developers for the great support!
Guido

flatmax, 2019-10-09:
Great!

Did you have to do anything strange to get it to work with the Pi 4 on 
your crux-arm OS ? I hadn't heard of CRUX  - thanks for the heads up.

keygee, 2019-10-10:
Hi Flatmax
CRUX is the predecessor of arch-linux. CRUX is source based, arch-linux is packed based but very similar...
After connecting the card to a PI2 with an old system (about 2016..... gcc 5.3.0!), the kernel did not see it.
Then I copied boot from https://github.com/raspberrypi/firmware/tree/master/boot to my /dev/mmcblk0p1, modules from https://github.com/raspberrypi/firmware/tree/master/modules to my /lib/modules/ and https://github.com/raspberrypi/firmware/tree/master/hardfp/opt/vc to my /opt/ (not sure it was necessary).
After reboot, the card worked great with your /etc/asound.conf.
Then I inserted the same sdcard to a PI4 and everything worked as expected (PI4 is too hot for my pourposes!), inserted to a PI3 and worked great as well.....

I'm only having problem with a custom asound.conf that I use with MPD to run multiple instances with separate out (after about a day of playing, MPD started to run pieces of audio (about 100ms) in loop. Maybe today I solved "merging" your and mine asound.conf....
If you or somebody are interested to it I'm happy to share.

Thank you again for the great work on it.

Ciao

Guido

p.s. Flatmax could you please contact me? I bought mine from audioinjector.net store via Paypal, but your email address is unreachable. We are interested to buy it int stock (if I found a solution with asound.conf! ;-) )

Ready2Burst, 2019-10-27:
Where can I buy these boards? I’ll need around a dozen of them.

flatmax, 2019-10-28:
shop.audioinjector.net

That is one location.

Amazon.com that is another.

Contact matt@ audio injector

dot net

That is another !

HinTak, 2020-01-07:
I needed to do something similar to
https://github.com/Audio-Injector/Octo/issues/14#issuecomment-475045553
to get it detected in buster ; stretch is easier.

elk-love, 2020-03-03:
I also tried the Octo on the Pi 4 with Buster and it works fine if I do input or output, but if I try to use both, the channels (almost always just the output channels) get assigned incorrectly. I have simple C ALSA code that just tries to copy 6 inputs to 6 outputs. Usually what ends up happening is that the audio output ends up being shifted over by two channels so that nothing comes out of the first two channels. Occasionally it will use the correct channels, but this is rare.

Also, when starting up, I usually have to do the [modprobe fix ](https://github.com/Audio-Injector/Octo/issues/14#issuecomment-475045553) to detect the Octo board.

Is there any example of working C code to just pass the inputs to the outputs?

HinTak, 2020-03-03:
But it is 6-in 8-out - you need to have two out empty anyway!

flatmax, 2020-03-03:
Indeed !

Hey elk-love, can you link us to your code example for full duplex processing ? 

Here is a full duplex example, it should be tweakable for many channels : 
https://github.com/flatmax/gtkiostream/blob/master/test/ALSAFullDuplexTest.C
It is based on this ALSAFullDuplex C++ class :
https://github.com/flatmax/gtkiostream/blob/master/include/ALSA/FullDuplex.H

elk-love, 2020-03-04:
This is what I'm trying to do. It works sometimes, but usually the channels are shifted. And I would like to eventually use all 6 inputs and 8 outputs, so I can't just leave two empty even if it was consistent in the amount that it is shifted. I will try to use the gtkiostream example that you linked to see if that works for me. Thanks!

`/*
 * Audio in/out using ALSA API and libasound.
 *
 * Compile:
 * $ gcc -o passthrough passthrough.c -lasound -lm
 *
 * Usage:
 * $ ./passthrough
 *
 */

#include <alsa/asoundlib.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <math.h>

#define PCM_DEVICE_IN "anyChannelCount"
#define PCM_DEVICE_OUT "anyChannelCount"

#define AUDIO_CH_IN        6
#define AUDIO_CH_OUT       8
#define FRAMES_PER_PERIOD  256
#define PCM_FORMAT         SND_PCM_FORMAT_S16_LE
#define SAMPLE_RATE        48000
#define NUM_FRAGMENTS       2

int audio_init(snd_pcm_t ** pcmp,
    const char * name,
    snd_pcm_stream_t stream,
    snd_pcm_access_t access,
    snd_pcm_format_t format,
    unsigned int channels,
    unsigned int rate
)
{
    int err;
    snd_pcm_hw_params_t *hw_params;
    
    if ((err = snd_pcm_open (pcmp, name, stream, 0)) < 0)
    {
        fprintf (stderr, "cannot open audio device %s (%s)\n", PCM_DEVICE_IN, snd_strerror(err));
        exit (1);
    }

    if ((err = snd_pcm_hw_params_malloc (&hw_params)) < 0)
    {
        fprintf (stderr, "cannot allocate hardware parameter structure (%s)\n", snd_strerror(err));
        exit (1);
    }

    if ((err = snd_pcm_hw_params_any (*pcmp, hw_params)) < 0)
    {
        fprintf (stderr, "cannot initialize hardware parameter structure (%s)\n", snd_strerror(err));
        exit (1);
    }

    if ((err = snd_pcm_hw_params_set_access (*pcmp, hw_params, access)) < 0)
    {
        fprintf (stderr, "cannot set access type (%s)\n", snd_strerror(err));
        exit (1);
    }

    if ((err = snd_pcm_hw_params_set_format (*pcmp, hw_params, format)) < 0)
    {
        fprintf (stderr, "cannot set sample format (%s)\n", snd_strerror(err));
        exit (1);
    }
[... 142 lines cut ...]
                // convert float back to integer
                buff_out[ch_n][n] = (int16_t)(buff_out_f[ch_n][n] * (float)pow(2, (bits_per_sample-1)));
            }
        }
        err = snd_pcm_wait(pcm_handle_out, 500);
        if (err == 0)
        {
            printf("output timed out\n");
        }
        err = snd_pcm_writen(pcm_handle_out, (void**)buff_out, frames_per_period);
        if (err == -EPIPE)
        {
            printf("PIPE");
            snd_pcm_prepare(pcm_handle_out);
        }
        else if (err < 0)
        {
            printf("ERROR. Can't write to PCM device. %s\n", snd_strerror(err));
        }
    }

    return 0;
}

`

flatmax, 2020-03-04:
Anytime.

If you want help with the example I have sent to you, a better 
communication medium for that may be a new email thread here :

https://lists.audioinjector.net/mailman/listinfo/people

I am shifting to discussions on that mailman email list as it is better 
for interaction in some ways.

elk-love, 2020-03-04:
I was able to install gtkiostream using the [Tutorial 0](http://forum.audioinjector.net/viewtopic.php?f=10&t=2982). I changed the deviceName to "anyChannelCount" and change ch to be 6. I compiled it with:
g++ -o ALSAFullDuplexTest ALSAFullDuplexTest.C `pkg-config --cflags --libs gtkIOStream

When i ran it, I got similar results to my code; the input to output channels were inconsistent in what was assigned. Does this code work for you on a Raspberry Pi 4 running Buster? If so, I'm baffled as to how my setup is different.

flatmax, 2020-03-04:
I recommend changing ch to 8.

There are always 8 channels in capture and playback. In capture 2 of 
these channels will always be 0.

elk-love, 2020-03-04:
In ALSAFullDuplexTest.C, I tried changing ch to 8 and that didn't change anything. The channels are still inconsistently assigned. I also tried changing the input channels to be 8 in the ALSA code that I wrote and it also didn't change anything. Any other ideas?

By the way, I realized that I left off the reverse single quote mark at the end of what I said I used for compiling. Just in case anyone is trying to duplicate this, here's what it should have been:
g++ -o ALSAFullDuplexTest ALSAFullDuplexTest.C `pkg-config --cflags --libs gtkIOStream`

elk-love, 2020-03-04:
Ok. I found one more piece of info that is very interesting. I just realized that the ALSAFullDuplex  program seems to work the first time that I run it after I reboot the Pi. If I ctrl-C out of the program and then run it again, I usually don't get the correct channel mapping. I have to do a "sudo reboot" to get back to a state where the channels are correct. Does this help narrow down the issue?

Development would be very slow if I had to reboot after every change to the code. Is there any way to get the system back to the state it is in after a reboot? If there was something that I could do when my code starts, that would be very helpful.

flatmax, 2020-03-04:
What happens when you wait some time and run the program ?

It is likely you have to wait for the clock to stop before you restart.

elk-love, 2020-03-04:
How long do I need to wait? I tried waiting one minute, but that didn't help. I tried putting a scope on the I2S bit clock and the LR clock, I see them stop immediately after I hit Ctrl-C, I don't see them go on for any discernible length of time.

flatmax, 2020-03-04:
OK - good to know.

I can't see that it will help, however it may be worth a try, there is a 
way to load the device tree overlay with the non stop clocks option. I 
can't see that this will help, but just good to exhaust all options :

dtoverlay=audioinjector-addons,non-stop-clocks=true

As I said, I don't think this will fix your problem. I would expect no 
channel swapping once the clocks have stopped.

elk-love, 2020-03-04:
Adding the non-stop-clocks=true didn't affect the channel shifting the second time that I run ALSAFullDuplex. The I2S clocks, however, now continue after I Ctrl-C out of the program.

flatmax, 2020-03-04:
OK, I didn't expect the non-stop-clocks to help.

I haven't experienced the problems you are having. Typically when the 
clocks stop, everything starts afresh. It should be the equivalent of a 
cold start.

Rather then using an ALSA plugin (anyChannelCount) what happens if you 
use the actual hardware hw:0 ? You will have to use 8 channels in this mode.

elk-love, 2020-03-04:
There was no difference when I use hw:0 vs anyChannelCount. It still works the first time only. I tried it both with and without the "non-stop-clocks=true" in /boot/config.txt

flatmax, 2020-03-05:
OK -  Personally I haven't experienced this channel swapping problem 
with restarts. I will have to sit down and test with an octo and the 
latest raspbian when I am not working remotely.

elk-love, 2020-03-10:
I have one more piece of information. I just got a Raspberry Pi 3B+, so I was able to test it with my Octo board, with pretty close to the same results that i got with the Pi 4. 

First I tried it booting Buster. The Octo was not detected whenever I rebooted, so I always had to do the modprobe fix. After detecting the Octo, I tried the gtkiostream ALSAFullDuplexTest. It passed audio through correctly the first time, but if I used Ctrl-C to quit the program and then ran ALSAFullDuplexTest again, the channels were shifted. If I then did a sudo reboot, did the modprobe fix, and then ran ALSAFullDuplexTest again, it would pass audio through correctly.

I then tried booting Stretch. Unlike Buster, the Octo was always detected correctly when booting Stretch. I ran gtkiostream ALSAFullDuplexTest.  It passed audio through correctly the first time, but if I used Ctrl-C to quit the program and then ran ALSAFullDuplexTest again, the channels were shifted. If I then did a sudo reboot, and then ran ALSAFullDuplexTest again, it would pass audio through correctly.

Could there be a problem with my particular Octo board? On the top silkscreen it is marked "v1.16". Is that the latest rev? Is there any other information that I can do to chase down this issue?

flatmax, 2020-03-10:
I don't think it is a problem with the hardware, I think that the 
software has changed somewhere.

Most likely the DMA subsystem or the I2S driver for the Pi.

HinTak, 2020-03-10:
This is the 3rd time this info is mentioned : https://github.com/Audio-Injector/Octo/issues/14#issuecomment-475045553 - recent kernel changes affected device probing and detection and, if you are using a recent kernel (e.g. Those shipped with Buster, instead of the older ones shipped with Stretch), you need the modprobe thing. 

Btw, I don't know why that issue is closed - it should be closed if a new release works by default...

flatmax, 2020-03-10:
OK - thanks @HinTak  - that is a good fix for now.

I disable the panel and reboot without the desktop, then I can reload the modules as mentioned here :
https://github.com/Audio-Injector/Octo/issues/14#issuecomment-475045553

```
sudo modprobe -r snd_soc_audioinjector_octo_soundcard
sudo modprobe -r snd_soc_cs42xx8_i2c
sudo modprobe -r snd_soc_cs42xx8
sudo modprobe snd_soc_cs42xx8
sudo modprobe snd_soc_cs42xx8_i2c
sudo modprobe snd_soc_audioinjector_octo_soundcard
```

The channel sync should be good after reloading. Note : if the following hangs :
`sudo modprobe -r snd_soc_audioinjector_octo_soundcard`
it is because the audio applet in the lxpanel is locking the removal of the sound card driver. You need to stop or unload that applet. I simply used raspi-config to boot without the desktop for now.

@HinTak are you aware of a way to stop the lxpanel audio applet ?

closed 2020-03-10 by flatmax

elk-love, 2020-03-11:
I looked at the comments that you added to https://github.com/raspberrypi/linux/issues/3496 and I think that it doesn't represent the results that I found. On both Buster and Stretch, I ended up with channels being shifted if the ALSAFullDuplexTest is killed and then restarted. The only different behavior between the two is that Stretch is always able to detect the Octo on startup, whereas with Buster, I have to do the modprobe fix.

flatmax, 2020-03-11:
Have you tried removing the modules and modprobing them again between runs ?
That resyncs the channels.

elk-love, 2020-03-11:
On the Pi 3B+ or the Pi 4 and Buster, I tried removing the modules and modprobing them again. That didn't help. The channels remained shifted.

On the Pi 3B+ and Stretch, I couldn't remove snd_soc_audioinjector_octo_soundcard, because it says "Device or Resource busy"

elk-love, 2020-03-11:
I was able to remove snd_soc_audioinjector_octo_soundcard from the 3B+ with Stretch after I did a "killall lxpanel". However, after using modprobe to remove and reload the drivers didn't help to unscramble the channels. 

I don't believe that I'm the only user of the Octo that is trying to use both the inputs and outputs simultaneously. How are other people getting this to work?

nettings, 2020-03-23:
For the record, I can report the same: on a Pi 4 with Buster, the device is not detected at boot time. Modprobe fix is required.

Mohawi, 2020-04-12:
@HinTak The issue #14 was closed before buster was released. With stretch, the issue was solved. Maybe the fix for raspbian stretch was not ported to buster 😕 
Since I ran in the same problems with the rpi4 / buster again, I'm wondering why this issue is closed now, since it is not fixed within the driver. The workaround exists though, but you have to dig deep to find it if you don't already know about it...

@elk-love have a look at https://github.com/Audio-Injector/Octo/issues/1#issuecomment-455135061 The issue with the shifted channels can be worked around with using jackd. As long as jackd is running, I have no shifts detected (having the octo opened via jackd with 96kHz). 

The only problem I recently ran into is the same as described in here: https://github.com/Audio-Injector/Octo/issues/8#issuecomment-603149992
But this only happens (as far as I witnessed) at high cpu load...

Under normal conditions I haven't witnessed high cpu load on my rpi 4 so far.
I use the octo as soundcard for a multiroom audio server running snapcast and a multiple mopidy instances. The octo inputs record from analog signals (audio from my TV, my vinyl player and cd player). I mirror the three inputs to three of the outputs (for having low latency if needed [especially for the TV audio]) and use the fourth output as an audio sink for the snapcast client (for having multiroom audio with higher latency if needed). My average cpu load is between 170% and 200% which - for this amount of manual copying and streaming - is ok, I guess...