Audio-Injector/Octo #21: IO problems?...

edsut, 2019-10-08:
Hi,
I am getting started with an Octo/RCA board.  I posted a question to the forum but then noticed the "[For support go to github](http://forum.audioinjector.net/viewtopic.php?f=5&t=5068)" post, so here I am...
I walked through the "[Easy automatic setup](http://forum.audioinjector.net/viewtopic.php?f=9&t=1222)", then found I had to create the /etc/modprobe.d/audioinjector-octo.conf file (following instructions in "[Octo fails to register...](http://forum.audioinjector.net/viewtopic.php?f=9&t=5746)" post on forum.   That gave me good results with aplay -l and arecord -l.  So far so good...

After that I just wanted to test all inputs and outputs.  I found that output channel 8 has a much lower level than the other channels.  This was tested with an 8-channel .wav file as well as some raw ALSA code that pushes out a sine wave to all 8 channels.  Same result in both cases: good signal for channels 1-7, much lower level for channel 8.
Then on the input side I have not been able to get any signal in.  I tried simply routing one input channel to one output channel (using basic ALSA C code that works fine for other devices); then also tried Audacity.  So far, nothing on the input.  I tried a few single-ended mics, same result.  

Note, that I am using a Pi4.

So... 2 questions:

1. Anyone having issues with 8 channels of output?
2. Do I need a preamp to use a simple single-ended microphone as input?

Thanks

flatmax, 2019-10-09:
You must have a very early manufacture of the Octo - is that right ? 
There was a problem which was fixed in later manufactures. Two resistors 
on the Octo break out board were switched in manufacture.

We can organise for a new output breakout board to be sent to you.

edsut, 2019-10-10:
I just bought it...
RCAout: V1.5; RCAin: V1.6; Octo: V1.16
Does this resolve the input problem or output or both?
Thanks

edsut, 2019-10-10:
If you can tell me what has to be changed, I can probably do it myself.

flatmax, 2019-10-10:
I found the old forum post on this topic. Hope that helps.
http://forum.audioinjector.net/viewtopic.php?f=9&t=3080

For the microphone input question - I think you will need a mic preamp. What type of microphones are you using ?

ktb83, 2020-02-16:
Hi @flatmax,

I also have an original RCA breakout board with the swapped resistors. I don't have high quality soldering equipment. Would it be possible to get a replacement?

Thank you,
Kyle

flatmax, 2020-02-16:
If you can get me your email address, I'll chat with you offline.

ktb83, 2020-02-16:
> If you can get me your email address, I'll chat with you offline.
> […](#)

ktb6915@gmail.com

AuntySaffron, 2021-07-02:
Hi
I also have an issue with the Input card. Aplayer _l and -L show the cards have been recognized and both the Inputs and Outputs are selectable via the Speaker icon. I have selected Duplex mode so both Inputs and Outputs should work.

Most of the outputs work. All but 1 so that is an issue however none of the inputs seem to be working at all. I have tried inputting from different sources but the system is silent
It is an RPI 4B and running Version 10 Buster and firmware dated 29/6/2021 16:03:53

Cheers for any assistance