Audio-Injector/Octo #22: Preamp for condenser mics and alsa mixer settings?

KarlssonCatharina, 2019-11-27:
Hi! 

I am in the processing of attempting to use a raspberry pi with the octo sound card to create a beam forming set up = 6 condenser microphones (Primo EM172's if it makes any difference). 

I get it to record on start up (after a five minute delay so we can get away from the set up) for an hour and split the recordings the way it needs to be done. But there seem to be zero sound in the wav files. From previous issues raised there seem to be a need for an amplifier and phantom power - however, I have no clue which one works with both the RPi and the octo sound card?  

I also wonder what the alsa mixer settings are supposed to be? For our one-channel recordings they have to be changed and I wonder if there's anything specific that needs to be done with them for six channel recordings? There are quite a lot of options for this sound card so I feel pretty lost with what each specific one do. I assume the F4- capture option should be at 100% but for single channel under F3 - playback controls the speaker should be at 100% and the mic at mute, does anything similar need to be done for the octo card? 

I have an urgent need to count frogs so would be very grateful if anyone could give me any direction on at least which preamp to get. 

Best Regards
C

nettings, 2020-03-23:
@KarlssonCatharina, I don't know your mics, but typically even permanently polarized condenser mics need phantom or line power to feed the little impedance converter that is in the capsule.
Also, the Octo's standard input board very likely doesn't have enough gain for those soft source signals. So you will need six external mic preamps or make a custom breakout board...

flatmax, 2020-03-23:
That seems right.
The Primo EM172 is an electret microphone and requires around 1.5 to 2V mic bias power.

My recommendation would be to create an active preamplifier on the board which includes a microphone bias line. The preamplifier may need as much as 30 dB of gain for those types of microphones.

You could use something like a 2.5V LDO regulator in series with a 2.2 kOhm resistor to create the microphone bias. Something like this :
https://i.stack.imgur.com/FcKzs.png
There is an AC coupling capacitor there so that the DC mic. bias voltage doesn't hit the codec's ADC.

You will also want to make some sort of preamplifier using an operational amplifier.