Audio-Injector/Octo #26: Examples?

djcobylevy, 2020-03-24:
hi,
I just purchased the Octo,
and so far - not so good.

I'm trying to record a single channels to different files using different instances of sox,
but i cannot seem to make it work.

Is it possible to get some examples?

Single channels recording,
multiple channels recording,

single channel playback,
multiple channel playback.

i think other people are looking for this information beside me.
it would be very helpful if you include some examples for us,

after all - we did believe in this product and purchased it.

thank you.

flatmax, 2020-03-24:
This is a good one for the mailing list :

https://lists.audioinjector.net/mailman/listinfo/people

flatmax, 2020-03-26:
This was answered on the mailing list here :
https://lists.audioinjector.net/pipermail/people/2020-March/000027.html
This is also relevant 
https://lists.audioinjector.net/pipermail/people/2020-March/000028.html