Audio-Injector/Octo #28: Audio pops on playback

fkwp, 2020-04-17:
Hi, 

every time the playback device is opened for playback a loud pop sound occurs. If the cadence between closing and opening the audio device is small (in the range of seconds) the pop sound does not occur. 

As workaround I use the the dmix plugin and  `aplay --rate 44100 --format=S24_LE /dev/zero` to keep the plackback device open.

Is there any chance to prevent the audio pop?

flatmax, 2020-04-23:
>
> Hi,
>
> every time the playback device is opened for playback a loud pop sound 
> occurs. If the cadence between closing and opening the audio device is 
> small (in the range of seconds) the pop sound does not occur.
This sounds right to me. Typically if the stop/start time is short, the 
DAPM subsystem doesn't shut down the audio clocks and bias voltage. The 
bias voltage shutdown will be causing the pops. On the ultra people 
found a way to stop the bias voltage from shutting down ... they enabled 
digital loopback. Not sure what the work around would be here.

> As workaround I use the the dmix plugin and |aplay --rate 44100 
> --format=S24_LE /dev/zero| to keep the plackback device open.
That is a good idea. If you use -Ddmix then you can open the output for 
playback with other applications at the same time.

> Is there any chance to prevent the audio pop?

Need some creative thinking here. If we do an Octo v2, I plan to use a 
new approach which will stop the popping problem.

fkwp, 2020-04-29:
> This sounds right to me. Typically if the stop/start time is short, the DAPM subsystem doesn't shut down the audio clocks and bias voltage. The bias voltage shutdown will be causing the pops. On the [...]

Do you have a hint where to research for the digital loopback? 

> Need some creative thinking here. If we do an Octo v2, I plan to use a new approach which will stop the popping problem.

I will be one of the first customers for the octo v2 :-)

flatmax, 2020-06-25:
Can you please tell us your use case for the card ?

fkwp, 2020-06-28:
> Can you please tell us your use case for the card ?
> […](#)

I am using the card as hi-res audio player (mpd) and for streaming (librespot, shairport). Especially using mpd it would be really nice to have exclusive access to the sound card since the library contains 44.1kHz, 48 kHz @ 16, 24 bit @ 2, 4, 5, 6 channels.

WangElectronics, 2021-09-29:
I have this popping problem as well. I work with pd and wonder how to keep the playback device open.