Audio-Injector/Octo #32: Not functioning at all.

markwinger, 2020-11-14:
I installed a Octo on a rpi 4.  I first installed manually to no success so I tried the automatic install.  That seem completed ok.

But, it is not working.  I have mixer program that was working on this pi with the audio injector stereo card very well.

But the software shows an alsa device called anyChannelCount. But if I try to use it I get an error trying to open the device and it says no channels.  The config.txt file looks correct as well as the .asoundrc.

So I tried running speaker-test.  here is the output:
pi@raspberrypi:~ $ speaker-test

speaker-test 1.1.8

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
Playback open error: -2,No such file or directory
pi@raspberrypi:~ $

So what is wrong with the setup

flatmax, 2020-11-15:
Let us try to diagnose the problem.

Does the command dmesg print out any information about finding the octo ?

Does "aplay -l" list the device ?

markwinger, 2020-11-15:
Dmesg shows nothing with audio injector.

[dms.txt](https://github.com/Audio-Injector/Octo/files/5541764/dms.txt)

No, aplay -l does not list the device.

markwinger, 2020-11-15:
Note, I have a dsi lcd display connected if it matters.

markwinger, 2020-11-15:
I flashed a fresh sd card, and installed and it works.  I think the problem was due to using gpio.  gpio 23 and 24 were set to input.  Somehow I got the default on boot to set them as input with pull ups but I could not get that back to where I started.  So I simply flashed a new card and it works.  I can use other gpio pins, so I will move them and see how things go from there.

markwinger, 2020-11-15:
Now I've tested and things work great until I shut down my ap.  Then I must reboot to get the audio working again.  

With the stereo audio injector something similar happened and running 
alsactl --file /usr/share/doc/audioInjector/asound.state.RCA.thru.test restore

would resolve it.  Is there a similar way to reset the octo?

flatmax, 2020-11-15:
Good going, solving the previous problem.

What symptoms do you see when you say it isn't working after you 
shutdown your app ?

Matt

markwinger, 2020-11-15:
My ap is a mixer with using all inputs and outputs.  After I terminate the ap and restart, I can get input and output to work.  It appears that input will work and I get no output.  But if I test the output separate it seems to work.  

I tried switching different devices, "default alsa","direct sample mixing","..no conversion" and sometimes I can get it to work again. 

Which device to you recommend?  
(I have not setup the high level alsa config, I don't seem to need it.)

markwinger, 2020-11-15:
I have not found a pattern to how this works, but it does not always fail to work after stopping it.  If I stop and and restart it repeatedly, sometimes it starts fine.  

As I said earlier, I had this problem with the stereo card, but adding the alsctl command to a startup script, it always work.

flatmax, 2020-11-16:
Ok - one way is to rmmod and modprobe liks this :

https://github.com/Audio-Injector/Octo/issues/20#issuecomment-597364408

However I think we need to tighten up the clock management in the 
driver. Have you tried launching the driver with the "non_stop_clocks" 
option with the device tree in config.txt

It does this in the driver :

https://github.com/raspberrypi/linux/blob/rpi-5.4.y/sound/soc/bcm/audioinjector-octo-soundcard.c#L144

Like this :

dtoverlay=audioinjector-addons,non_stop_clocks

Matt

markwinger, 2020-11-16:
Adding non_stop_clocks seems to have fixed the restart problem.

The only issue I see now if a very loud pop when booting, and every time I start my program.  Sounds like a dc offset into a coupling cap(I have not seen the schematic so this is simply my guess by listening(I'm an OLD analog guy).  If I could see the schematics I may be able to help correct that).  

Other than the pop, things are looking (sounding) good.

markwinger, 2020-11-16:
New Problem.  Now I keep getting an error when I configure input.  The error is device didn't start.  I cant make it work any more.

If I run the commands you specified in commend #20, my software can no longer see the card at all and I need to reboot.

markwinger, 2020-11-16:
I'm dead in the water now.  Just incase the OS was corrupted, I started with a fresh install again but I cannot get audio input device to start.

Just before this happened, I was getting lots of pops in the audio, and I noticed the rpi temp was up to 75C, so I figured the processors were being temperature throttled.  So I shut down to let it cool.  When I rebooted and hour later I could not longer get the audio to work.  I think either the rpi failed or the Octo has failed.  I don't have a spare rpi4 to test on so I can't confirm.

flatmax, 2020-11-16:
Sorry to hear that.

I wonder if some hardware was accidentally shorted somewhere ? Possibly 
creating extra heat.

markwinger, 2020-11-16:
The software runs the pi pretty much full tilt.  I have not put a fan on it so 75C is not surprising and of course the Octo will kind of hold the heat in without a fan.  With a fan I should be able to keep it below 50C.

markwinger, 2020-11-17:
Dmesg show this:
 167.964335] logitech-hidpp-device 0003:046D:401B.0004: HID++ 2.0 device connected.
[  180.877159] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  180.880720] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  186.441871] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  186.447869] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  274.277384] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  274.293703] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  278.333092] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  278.349660] bcm2835-i2s fe203000.i2s: I2S SYNC error!

I suspect the pi i2s failed.  I will get a replacement pi (and a fan) tomorrow and see if the problem is resolved.

markwinger, 2020-11-17:
I got a new rpi. Same issue.  It appears that the rpi was not the problem but the Octo hardware has failed.  

I'm sure the Octo was damaged by heat.  That temp of 75C was the pi core processor temp and it is rated for max of 85C and the firmware was controlling properly.  

How do I get a replacement? I've had it on 5 days, it worked for only 2.   Should I return to Amazon and reorder?

markwinger, 2020-11-17:
Amazon is sending a replacement.  I will let you know how the replacement goes.

markwinger, 2020-11-18:
Update: have not received replacement you.  But occasionally it will start to work again for a while then after a reboot it may fail to start.  There appears to be and intermittent hardware problem.

Also, up until it failed I never had a problem with pops/clicks in the audio.  Now I get them a lot of them.  I also now have the sync problem where the input/output channels may change each time I start.  To run the mod probe command in #20, I need to terminate the lxpanel, run it then log back in or it will hang.  (I need the desktop for my ap).  Can I get lxpanel to release it without killing it?

flatmax, 2020-11-19:
Right the lxpanel thing is still an issue. You don't need to kill it !

Here is an older post where I was disabling the lxpanel volumealsa 
control which was locking.

https://raspberrypi.stackexchange.com/questions/70922/configure-alsa-sound-card

More recently I have used the following :

sudo sed -i "s/mount_on_startup=1/mount_on_startup=0/" 
/etc/xdg/pcmanfm/LXDE-pi/pcmanfm.conf

Does this command still work ?

Not sure what the clicks and pops are from - apologies.

Matt

markwinger, 2020-11-19:
Still waiting for the replacement, but I determined something different.

If the card is recognized when booting, it will sometimes wor

markwinger, 2020-11-19:
I am trying to determine if my use of gpio pins is causing problems.  I found a post indicating gpio 5,18-24 and 27 were used.  I am using 1,7,8 and plan to use 9 and 10 also.  Can you confirm there is not a conflict?

markwinger, 2020-11-23:
You asked if sudo sed -i "s/mount_on_startup=1/mount_on_startup=0/"
/etc/xdg/pcmanfm/LXDE-pi/pcmanfm.conf still worked.  It executed, and sets mount_on_startup to 0.  But running the the modprobe commands still hangs.  I also tried setting automount=0.   That did not work either.

A week ago, the card appeared to have failed completely.  I could not get it work at all.  On boot, aplay -L would sometimes say no audio cards.  Sometimes it wold work.  But then starting my mixer would error when I tried to configure input with device failed to start.  I tried for hours with no luck.  Called amazon to ship a new card.

Then next day it booted, and worked, for a while.  If I got it working, all would be fine until I stopped and restarted my mxer or rebooted.  Then sometimes it would work, other times no.

The card came and it behaves the same way, but I have not seen the aplay -l no audio card found error for a while on either card.

Now, it seems to boot with aplay listing correct at each boot.

As of now I have startup script that terminates lxpanel, does the modprobe commands, restarts the lxpanel in background then starts the mixer.  This works most of the time.  Pops and clicks seem to be gone now.  

Any ideas on why I cannot get it to work every time?

markwinger, 2020-11-23:
One more wierd one.

I powered up and could not get the card to start.  dmesg output:
[  321.880935] cs42xx8 1-0048: 1-0048 supply VA not found, using dummy regulator
[  321.880988] cs42xx8 1-0048: 1-0048 supply VD not found, using dummy regulator
[  321.881030] cs42xx8 1-0048: 1-0048 supply VLS not found, using dummy regulator
[  321.881075] cs42xx8 1-0048: 1-0048 supply VLC not found, using dummy regulator
[  321.903378] cs42xx8 1-0048: found device, revision 4
[  323.483999] audioinjector-octo soc:sound: cs42448 <-> fe203000.i2s mapping ok
[  328.895900] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  328.925454] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  328.932824] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  334.417213] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  334.419907] bcm2835-i2s fe203000.i2s: I2S SYNC error!
I would see the I2S sync error every time I started the mixer.   I then disconnect my rotary encoder, and reboot.  Things are back to description above.   I can reconnect the encoder and all works.

It appears that when the encoder(and fan) are connected at power up, it fails.  Disconnect, then power up it will work.  However, I still some sync errors:
 561.044669] cs42xx8 1-0048: 1-0048 supply VA not found, using dummy regulator
[  561.044723] cs42xx8 1-0048: 1-0048 supply VD not found, using dummy regulator
[  561.044767] cs42xx8 1-0048: 1-0048 supply VLS not found, using dummy regulator
[  561.044810] cs42xx8 1-0048: 1-0048 supply VLC not found, using dummy regulator
[  561.067483] cs42xx8 1-0048: found device, revision 4
[  562.648049] audioinjector-octo soc:sound: cs42448 <-> fe203000.i2s mapping ok
[  568.004154] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  568.009457] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  568.016889] bcm2835-i2s fe203000.i2s: I2S SYNC error!

Notice I only see 3 errors but when failing I have 5 errors.

Fan is connected to 5V and ground.  Encoder is connected to gpio7 and gpio8, the push button to gpio12.

flatmax, 2020-11-23:
I just updated this issue which links to the used GPIO pins :

https://github.com/Audio-Injector/Octo/issues/25

In summary :
For I2S :
pin 12 : Bit clock (GPIO18)
pin 35: LR clock (GPIO19)
pin 38: Data in (GPIO20)
pint 40: Data out (GPIO21)

For codec controls :
pins : 13, 15, 16, 18, 29
GPIOs : 27, 22, 23, 24, 5

Yes, I think you are correct. It should be ok to use those GPIOs.

flatmax, 2020-11-23:
I feel a little out of my depth with the setup you have. Apologies.

markwinger, 2020-11-23:
The setup is not that complex.  I use an ap (i wrote in c++) that is a very complex mixer.  But that hardware setup is simple.  I have a single rotary encoder, with a push button.  Pressing brings up menus that can be navigated with rotary encoder.  This uses 3 gpio pins.  Then I can use 2 more gpio (not currently connected) that connect to a foot switch for enabling and disabling effects.  Thats it.

The software has been running on windows for years in a large touring sound system.  It can scale to any number of inputs/output/eq/effects etc.  So when I saw the octo I decided it was time to get it running on a pi for a small sound system for soloist, duo or trio.  The rotary encoder is part of scaling to a minimum display and external controls.

I am building preamps for it now.  If I can get around the boot/sync problems with the octo it should work great.  As of now, 6 channels in, 6 out (sub/mains/monitor) crossover, eq effects and it can run for hours with no dropouts.  I'm even running 1.2 ms latency and it looks like I could go lower.

markwinger, 2020-11-23:
Could the issue be noise on the i2c bus causing the synchronization issues?

markwinger, 2020-11-23:
I know I'm giving you lots of info but the hope is it will help.

I see the sync problem sometimes where channels are scrambled.  I have the modprobe commands as part of my mix start up script.  I have a stereo source connected to channels 1+2.  I just started it and 1+2 came in as 5+6.  Restarting they switched to 1+2.  I thought the modprobes were supposed to get the channels synchronized, but it does not seem to always work.

markwinger, 2020-11-24:
Right now the system is pretty stable, but the channel scramble is a real problem.  I sometimes have to start the mix 7 or 8 times before it starts with the channels right.

A common sequence is channel 5-6 mapped to 3-4 after boot.  Restarting gets the channels mapped 5-6 to 1-2.  Restarting again maps 5-6 to nothing.  The next restart may them map correctly.  The following is inlcuded in the start script before the executable is started:
sudo modprobe -r snd_soc_audioinjector_octo_soundcard
sudo modprobe -r snd_soc_cs42xx8_i2c
sudo modprobe -r snd_soc_cs42xx8
sudo modprobe snd_soc_cs42xx8
sudo modprobe snd_soc_cs42xx8_i2c
sudo modprobe snd_soc_audioinjector_octo_soundcard

In the thread about this back in April there was a discussion about adding a sync bit to the buffer to identify the channel.  It seems to me actually putting the channel number in each channel would be better.  Checking any channels number would tell you exactly how far out of synchronization you are.

**Is there any chance of implementing some kind of fix, soon?  Please?  I can help, maybe.**

markwinger, 2020-11-25:
I have a workaround for the channel scramble.   I simply increased the input count to 8, the look at the first data transfer and look for the empty frame.  From there I know which channels should go where.  My program has routing of inputs and outputs.   I simply used that empty frame index to modify my input routing map automatically at startup.  This works every time as long as sync never gets lost(I have never seen this happen once I'm running).

flatmax, 2020-12-10:
Yes, 
That is very similar to how the input descramble in software works. Because the card has 6 channels of inputs on an 8 channel bus, the input lines always have 2 channels in which the captured data is constantly the same number. Using this information the software un-scramble can work. Have a  look at this code here :
https://github.com/flatmax/gtkiostream/blob/master/test/ALSAFullDuplexMinScan.C#L51-L76

markwinger, 2020-12-10:
As I mentioned above, I implemented a workaround this in type of fix in my mixer.  But I have the output scramble issue as well.

After boot, the first time a start my program it always works.  But once I stop it it always comes up scrambled and I dont know how to identify the correct mapping to work around this problem.

markwinger, 2020-12-12:
I just realized I had the non-stop-clocks set up wrong.  I had non_stop_clocks instead.  I corrected this and now the outputs no longer scramble on restart.