Audio-Injector/Octo #35: Channels count non available

danvy82, 2021-03-14:
Hi,

I am having the following issue with the Octo:

`arecord --device=hw:0,0 --format S16_LE --rate 96000 -c 6 -d 10 --max-file-time=30 TestAudio_.wav
Recording WAVE 'TestAudio_.wav' : Signed 16 bit Little Endian, Rate 96000 Hz, Channels 6
arecord: set_params:1345: Channels count non available
`

I followed the [instructions ](http://forum.audioinjector.net/viewtopic.php@f=9&t=1222.html)from the forum. I noticed another similar [post ](http://forum.audioinjector.net/viewtopic.php@t=2659.html) where the where ".asoundrc" was modified by Pixel, but this seems not what is happening in my case. The values correspond to what is in the forum.

Here are the contents of the asound files and the output of arecord -L : [OctoIssue.txt](https://github.com/Audio-Injector/Octo/files/6137887/OctoIssue.txt)

Do you see what could be wrong? Thanks for any support.

flatmax, 2021-03-18:
I would like to remind github users, that we don't officially support here - search for audioinjector people. Use our email list.

When you do :
```
aplay -L
```
and
```
arecord -L
```
Do you see the anyChannelCount device in both cases ?

danvy82, 2021-03-20:
Thanks for getting back on this and I got it somehow working without specifying "--device=hw:0,0". I'll use the mailing list in case of other questions.

closed 2021-03-20 by danvy82