Audio-Injector/Octo #36: I/O Channel swapping using octo and SuperCollider

albertorizzo, 2021-03-27:
I would like to report channels swapping with full-duplex operations using the same audio application
I'm using Supercollider (and specifically the scsynth server), to process and receive/playback the signal in real-time at 96 kHz (the application deal with these tasks, no other parties involved).
Finally, I use jackd to route the audio buses of supercollider to the physical output of the octo soundcard.
Every time I try to make it work, I notice channel swapping both on input and output side (RCA).
The supercollider bus order is the same every time.
 I've read several topics on this problem, but it looks like this is something different.
As far as I understand, Supercollider audio server works synchronously.
I have the same problem using Audacity or Reaper with full-duplex operations.

Here's some additional info.

I'm currently using this line in my /boot/config.txt file:

dtoverlay=audioinjector-addons,non-stop-clocks=true

This is my .asoundrc file:

pcm.!default {
#       type hw
#       card 0
        type plug
        slave.pcm "anyChannelCount"
}

ctl.!default {
        type hw
        card 0
}

pcm.anyChannelCount {
    type route
    slave.pcm "hw:0"
    slave.channels 8;
    ttable {
           0.0 1
           1.1 1
           2.2 1
           3.3 1
           4.4 1
           5.5 1
           6.6 1
           7.7 1
    }
}

ctl.anyChannelCount {
    type hw;
    card 0;
}

gashalot, 2021-09-03:
I'll cross post this issue here, as I have also encountered the same issues. I do not see channel swapping as frequently as others note here, but it happens across reboots and long pauses between playing different sources, channels swap unpredictably. Setting the `non-stop-clocks` parameter seems to have no impact on the situation.

After a good amount of research, this may be related to hardware limitations with the device(s) themselves. Upstream issue linked below. Unclear if this can be fixed in software or not.

https://github.com/raspberrypi/linux/issues/2473

flatmax, 2021-09-03:
There are a few reported ways to fix this channel swapping problem. I think the best solution is to fix it in software.
This code here shows you how to reliably shift input data to always get the correct channel alignment :
https://github.com/flatmax/gtkiostream/blob/master/test/ALSAFullDuplexMinScan.C#L50-L70

A student at a university in China wanted to use the Octo with relaiable output as well and I guided him on how to apply the same channel shifts to the output as well. With a bit of tinkering they reported success.

The best solution would be to either have that channel alignment code in your software OR if you aren't implementing it in your software (using other people's software) then to have an ALSA plugin which performs the minimum input channel scanning technique.

FBStenBerge, 2022-07-01:
Could you please give a stepwise instruction how to have this code implemented so pure data has both the inputs and outputs aligned correctly?