Audio-Injector/Octo #40: Everytime i attach the card to the pi it crashes the pi

tednesseth, 2021-06-05:
I just got my pi up and running 
I tried to install the card to install the software
it crashes every time 
Pi wont get back online until i remove the soundcard.

Any help would be great

tednesseth, 2021-06-06:
Guys I get all the way to the end but every time i connect the hardware the pi wont boot.
please let me know what i need to do

thanks on advance

flatmax, 2021-06-06:
Do you mean that it will not boot at all ? Or that it boots but stops at a certain point?
Does the board get hot anywhere ?

tednesseth, 2021-06-06:
> PHISH ALERT! CHECK VALIDITY IF CLICKING, SHARING, RESPONDING

Hey man the board does not get hot at all

What happens is I get to the point where I’m going to power down in it and plug in the sound card and then power the module connected to the sound card and that’s where I can’t get a host name

THEODORE NESSETH | SR ARCHITECT
***@***.******@***.***>
M 737.210.2445<tel:737.210.2445> |
9600 Great Hills Trail, Suite 300E, Austin, TX 78759<x-apple-data-detectors://2>
e2open.com<http://e2open.com/> | Twitter |

tednesseth, 2021-06-06:
> Do you mean that it will not boot at all ? Or that it boots but stops at a certain point?
> Does the board get hot anywhere ?

@flatmax i guess its pretty how at the pi power supply
and hot under the cirrus logic chip

Also i should mention ( just in case) i am booting with the ethernet on the pi - was having no luck with wifi

flatmax, 2021-06-06:
ok - what happens on the screen when you plug in an HDMI connector ?

tnesseth1, 2021-06-06:
i am headless. i've been doing through my laptop.

flatmax, 2021-06-06:
Well, perhaps you can jack in and see if you can login with a keyboard ? Make more diagnoses ?

We quality control our hardware before it goes out. Having said that, occasionally some products defect after quality control, but they are small in number. Do you think you have a hardware defect on the board ?

tnesseth1, 2021-06-06:
I have no idea if I have a defect or not I bought it from Amazon a couple months ago just had a chance to get it going in the last couple days￼

flatmax, 2021-06-07:
ok - Do you have anything else plugged in ?

tnesseth1, 2021-06-07:
just the ethernet and power

tednesseth, 2021-06-18:
I'm trying the suggestions sent - will follow up

tednesseth, 2021-06-19:
I must either have a faulty octo card or raspberry pi. i have another pi arriving tmrw - i will follow the same process and if i stil cant boot after the pi after the octo card is on it has to be the octo card.  -can you confirm if one should be able to have the octo card pinned to the pi before any oct config takes place or will that also freeze the pi?

tednesseth, 2021-06-20:
i am now pretty convinced i have a bad sound card 

[image]

you can see mine is compromised

what do i need to do to get a replacement?

tednesseth, 2021-06-20:
I bought it on amazon and it looks like a hack job compared to the ones on the site

flatmax, 2021-06-20:
Lets take this issue off github. I'll try to contact you directly.

tednesseth, 2021-06-20:
***@***.***

delete
my
email after

thanks!

Lets take this issue off github. I'll try to contact you directly.

—

tnesseth1, 2021-06-21:
I got another octo.
im gonna get back on the merry-go-round

tednesseth, 2021-06-22:
closer

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: audioinjectoroc [audioinjector-octo-soundcard], device 0: AudioInject-HIFI cs42448-0 [AudioInject-HIFI cs42448-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

whats next and how do i test? is there a file i can test with handy?

tednesseth, 2021-06-23:
are you planning on getting back to me as discussed?
or can you direct me somewhere to troubleshoot?