Audio-Injector/Octo #42: Output channel Mixed

razvysme, 2021-06-21:
Hi, I am using and Octo on a RPi 4 with Patchbox OS, and sometimes it boots, the output channels are mixed up. What it seems is (like others) channel 7-8 become 1-2. 

For my particular application I only need 6 channels of audio output (currently to set 3-8).  I am using Pure-Data 0.5, running with GUI on @ startup. 

If I disable the driver and re-enable it(by turning on the DSP on/off), the sound cards works 50% of the time(first time I re-enable it, there is no sound, 2nd time channels are all mixed up). 

I have been reading about this issue, and it seems rather common with the OCTO, but the solution is not evident to me.

gashalot, 2021-09-03:
I'll cross post this issue here, as I have also encountered the same issues. I do not see channel swapping as frequently as others note here, but it happens across reboots and long pauses between playing different sources, channels swap unpredictably. Setting the `non-stop-clocks` parameter seems to have no impact on the situation.

After a good amount of research, this may be related to hardware limitations with the device(s) themselves. Upstream issue linked below. Unclear if this can be fixed in software or not.

https://github.com/raspberrypi/linux/issues/2473

flatmax, 2021-09-03:
I have a response to this problem here :
https://github.com/Audio-Injector/Octo/issues/36

The basic idea is to have a software fix which min samples in the input channel scanning.