Audio-Injector/Octo #45: Dead channel (3) on output driver

breaker-of-stone, 2021-07-07:
Hi Team.

We purchased an Octo through Amazon and have built/ configured according to the README.

Audio plays clearly on Ch 1,2,4,5,6,7,8 RAC connectors.
Ch3 RCA produces nothing.  Checked, and it is not shorted.
Ch3 audio is present on the 24 pin DIL cable + plug, so audio's getting produced by the Octo.

What would you suggest we do next?

P.S.  How can we quickly test audio input?  No clues given in the README.

flatmax, 2021-07-07:
Quickly test recording us arecord. If you want to ensure that channels 
are not swapped, then you may want to convert this code to something 
which outputs to an audio file :

https://github.com/flatmax/gtkiostream/blob/master/test/ALSAFullDuplexMinScan.C

Regarding hardware issues, please contact me off github.

Matt

breaker-of-stone, 2021-07-09:
Ok, I may be missing something obvious, but how?

breaker-of-stone, 2021-07-11:
Does this suffice as contact?

Have tried twatter, farcebook and the flatmax.com contact form without
result.

flatmax, 2021-07-12:
The best way to discuss is on the email list : https://lists.audioinjector.net/mailman/listinfo/people

VehasSady, 2021-09-07:
Hi !

I recently has a similar problem.
My channel 3 was ok until today (just 3 days before the installation lol) but since a few hours it doesdn't work (while the other channels are fonctionnals). Have you got an idea of my problem ?
Thank you so much !!

Véhas