Audio-Injector/Octo #52: Automated install only works with default username

TubularCorpration, 2022-05-08:
Just a heads up, the "error exit status 2" during the install that a few people have reported is happening because the package tries to locate .asoundrc in /home/pi/ and will not work when logged in with any other username.

I am using mine for a mostly-offline project with a fresh Raspberry Pi OS 64 bit, so it was easiest for me to reinstall the OS and use the default username and that seems to have worked, but it's not a good option for most people so it would be great to make the automatic install package detect or prompt for the correct home directory instead of assuming the default and needing manual installation otherwise.

toaniel, 2023-02-26:
This is correct. I had to reinstall the OS with user pi for it to work.

mardulas, 2023-05-19:
Yes, it irked me too... But you can modify the install after the fact (or manual install).
Also the installer script is not going to make it work anyway, as it has issues w. the new kernel (need to downgrade kernel).