Audio-Injector/Octo #64: Noise in all inputs with kernel 6

marclava, 2023-06-26:
Inputs are working with Linux kernel 5, but not with Linux kernel 6; there's only noise at maximum level in all 6 inputs.
Outputs are working.
I tried with recent versions of Raspian and Ubuntu.

This is a important issue because newer versions of RPI OSes will soon use Linux 6+, including those dedicated to audio.

KarelDrummer, 2023-09-27:
To be honest, even outputs don't work with Kernel (+modules) 6.x. I manually migrated to 5.15 and it works.

It's a quite bit disadvantage of Octo - very bad software support + maintanance :(

KarelDrummer, 2023-09-27:
Event it doesn't work with Kernel 5.10 for me. I was lucky, I'm able manually load saved kernel 5.15 with modules, and manuall deploy really works. If you will be interested for this Kernel files, let me know at:

karel.drummer@hotmail.com

sangibeaud, 2023-10-13:
The issue might be because linux kernel now a proper driver for cs42xx8, and can handle up to 8 channels "natively", while audio injector octo had a FPGA on the board re-clock signals to accommodate old drivers.

Works fine for me with :

           sudo rpi-update fab7796df0cf29f9563b507a59ce5b17d93e0390

which downgrades to kernel 5.10.y

Other kernel tags should work, if before 5.15 ish.
Sorry I don't know of a workaround to alter the driver and make it work in 6.x

Pete1162, 2026-10-05:
The issue has been solved: https://github.com/Audio-Injector/Octo/issues/74