Audio-Injector/Octo #65: Latest kernel working

deepMago, 2023-08-02:
For now I'm trying to use Octo with Kernel 4.19 on Buster.
I would like to know latest working Kernel

KarelDrummer, 2023-09-27:
It doesn't. With Kernel 5.15 works ok.

deepMago, 2023-09-28:
Thank you, it's a very useful information

michaelbbaker, 2023-12-11:
hi flatmax, I'm on buster (32bit) with rpi 4B.
I tried rpi-update pulls/5705. 
uname -r gives me: 6.1.61-v8+
The system recognizes the octo, but when I play audio via aplay, I just get strange digital noise on all the octo outputs. Happy to share any other system info that would help.

James-Lehman, 2023-12-13:
I just started with RPiOS 32bit lite (no desktop) Bullseye. Did the deb. First it didn't load correctly, so I made that sh script to remove and reload the kernel modules. That made it appear. But the same noise on all 8 channels with speaker-test. Tried the rpi-update pulls/5705. Same noise. :(

michaellucenkiw, 2024-01-21:
I am also getting just a blast of noise on all channels, no distinguishable audio....

flatmax, 2024-03-07:
Yes - confirmed - the drivers load in the newer kernel versions, but noise comes out. You still need to downgrade, until this problem is fixed.

HAve are look at this link : https://github.com/Audio-Injector/Octo/issues/66#issuecomment-1968919332

vanmanda, 2024-03-25:
I've been testing with firmwares.
As suggested by jgeskens the card works ok with firmware c145bdde6cf0a4096dc799a9067e05f64b463352, this corresponds to 5.15.65+.
You can upgrade to firmware f5c4fc199c8d8423cb427e509563737d1ac21f3c which is the last v5 5.15.92 and the card still works ok.

When running 5.15.65+ or 5.15.92, I always see this error:
[ 30.598040] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[ 30.745596] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[ 30.748138] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[ 30.753679] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[ 30.756243] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[ 30.988196] bcm2835-i2s fe203000.i2s: I2S SYNC error!
The system mostly survives this message, but sometimes it hangs after posting these messages:
[ 256.354900] Internal error: Oops: 96000005 [https://github.com/Audio-Injector/Octo/issues/1] PREEMPT SMP
[ 256.355320] Code: 8b1302b3 b25f7be0 d34cfe73 8b131813 (f9400660)

Once you upgrade to v6, you get distortion.
I see this error with v6:
[ 13.889229] audioinjector-octo soc:sound: ASoC: driver name too long 'audioinjector-octo-soundcard' -> 'audioinjector-o'

@vanmanda

Add a comment
Comment
 
Add your comment here...
 
Remember, contributions to this repository should follow our [GitHub Community Guidelines](https://docs.github.com/articles/github-community-guidelines).
Assignees
No one assigned
Labels
None yet
Projects
None yet
Milestone
No milestone
Development
No branches or pull requests

Notifications
Customize
You’re receiving notifications because you commented.
4 participants

0d0a, 2024-05-13:
any progress ? stereo soundcard works ok on a pi400 bookworm 64 bits. How about the octo soundcard?

Pete1162, 2026-10-05:
The problem has been solved: https://github.com/Audio-Injector/Octo/issues/74