Audio-Injector/Octo #66: Dec. 6th 2023 Octo Status please.

James-Lehman, 2023-12-06:
Hello everyone.

I ordered an Octo from the provided link to Australia China Cooperation Store and it took about 2.5 months to arrive. I have been trying for a couple of days now to get it to work on an RPi4 4gb. I've been reading everything I can find and I would like to get some information consolidated in one thread.

0. Is this device still supported?

1. Are there any Octo version issues to be aware of? Mine says:

Audio Inector.net
Octo sound card
PPI8 v1.16

2. Is there a combination of RPi model, OS image, exact steps to take, magic incantations, etc... that are known to always work?

3. Does the Octo work in 64 bit?

4. Will the Octo work on the new RPi5 or for that matter is there a way to connect it to any other Pi (Orange) as long as the pins are connected to the right stuff? If so, how and what?

5. Is there any information about installing the driver when it is not possible to use the provided deb file?

6. Is there a pin-out for the in, out ribbon headers?

Thank you.
James.

michaelbbaker, 2023-12-10:
Following!
The instructions on the readme don't work anymore (rpi 4b)
I'm slowly working my way through trying combinations of os and kernel, still using the .deb file linked here. 
I'll post if I get it working, but it would be really helpful if someone had a simple workflow for getting the octo functioning with one of the OS images available from raspberry pi (currently bookworm and bullseye 32 or 64bit). 
If a different OS image is required, which one, and ideally, where do we get it?

michaelbbaker, 2023-12-11:
Hi James, 
with the kernel version @flatmax [posted in one of the other issue threads](https://github.com/Audio-Injector/Octo/issues/65#issuecomment-1837811289), 
I can get to the weird digital noise point where you're at with Bullseye (32 and 64 bit versions). So it seems like this issue isn't specifically OS related. What I don't get is that when I use old kernel/os combinations that seemed to fix the noise problem for other users, I don't get any further.

James-Lehman, 2023-12-14:
I have two of these devices now. The first one got stuck in US Customs, so I ordered another one. They both do the same thing. They just make weird noises on all 8 channels.

jgeskens, 2024-02-28:
Downgrading the kernel worked for me, on the Bullseye Lite (legacy):

```
sudo rpi-update c145bdde6cf0a4096dc799a9067e05f64b463352
```

That said, the driver is outdated, it does not work anymore with newer kernels because you get the crackling sound.

James-Lehman, 2024-02-29:
Wow!

It took a bit of messing around, but I got the 32bit Bullseye version with the desktop and it works!

rpi-update c145bdde6cf0a4096dc799a9067e05f64b463352

Did the trick!

Thank you!

vanmanda, 2024-03-25:
I've been testing with firmwares. 
As suggested by jgeskens the card works ok with firmware c145bdde6cf0a4096dc799a9067e05f64b463352, this corresponds to 5.15.65+.
You can upgrade to firmware f5c4fc199c8d8423cb427e509563737d1ac21f3c which is the last v5 5.15.92 and the card still works ok. 

When running 5.15.65+ or 5.15.92, I always see this error:
[   30.598040] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[   30.745596] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[   30.748138] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[   30.753679] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[   30.756243] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[   30.988196] bcm2835-i2s fe203000.i2s: I2S SYNC error!
The system mostly survives this message, but sometimes it hangs after posting these messages:
[   256.354900] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[   256.355320] Code: 8b1302b3 b25f7be0 d34cfe73 8b131813 (f9400660)

Once you upgrade to v6, you get distortion. 
I see this error with v6:
[   13.889229] audioinjector-octo soc:sound: ASoC: driver name too long 'audioinjector-octo-soundcard' -> 'audioinjector-o'

vanmanda, 2024-03-25:
Additional log output:
[    7.262716] audioinjector-octo soc:sound: snd_soc_register_card failed (-517)
[    7.285344] uart-pl011 fe201000.serial: no DMA platform data
[    7.460191] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    7.736531] rpivid feb10000.codec: Device registered as /dev/video19
[    8.830764] audioinjector-octo soc:sound: snd_soc_register_card failed (-517)
[    8.940283] cs42xx8 1-0048: supply VA not found, using dummy regulator
[    8.940647] cs42xx8 1-0048: supply VD not found, using dummy regulator
[    8.940771] cs42xx8 1-0048: supply VLS not found, using dummy regulator
[    8.940876] cs42xx8 1-0048: supply VLC not found, using dummy regulator
[    8.957825] cs42xx8 1-0048: found device, revision 4
[    9.347541] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    9.348736] bcmgenet fd580000.ethernet eth0: Link is Down
[    9.419993] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.955061] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   10.366746] ------------[ cut here ]------------
[   10.366761] WARNING: CPU: 0 PID: 39 at sound/core/init.c:309 snd_card_init+0x34c/0x378 [snd]
[   10.366805] Modules linked in: snd_soc_cs42xx8_i2c snd_soc_cs42xx8 regmap_i2c overlay spidev rpivid_hevc(C) uinput bcm2835_codec(C) bcm2835_isp(C) brcmfmac bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops brcmutil snd_soc_audioinjector_octo_soundcard snd_soc_bcm2835_i2s videobuf2_v4l2 cfg80211 videobuf2_common snd_soc_core videodev raspberrypi_hwmon snd_compress snd_pcm_dmaengine rfkill snd_bcm2835(C) i2c_bcm2835 spi_bcm2835 snd_pcm vc_sm_cma(C) mc snd_timer snd uio_pdrv_genirq uio nvmem_rmem joydev hid_multitouch drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[   10.366907] CPU: 0 PID: 39 Comm: kworker/u8:1 Tainted: G         C         6.1.13-v8+ #1632
[   10.366913] Hardware name: Raspberry Pi Compute Module 4 Rev 1.1 (DT)
[   10.366917] Workqueue: events_unbound deferred_probe_work_func
[   10.366929] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   10.366934] pc : snd_card_init+0x34c/0x378 [snd]
[   10.366959] lr : snd_card_init+0xc4/0x378 [snd]
[   10.366982] sp : ffffffc00857b980
[   10.366985] x29: ffffffc00857b980 x28: 0000000000000000 x27: 0000000000000000
[   10.366995] x26: 0000000000000000 x25: ffffffddda665000 x24: 0000000000000000
[   10.367003] x23: 0000000000000000 x22: ffffffddda665088 x21: ffffff8100241c10
[   10.367012] x20: ffffffddda665d88 x19: ffffff81082eb000 x18: 0000000000000000
[   10.367020] x17: 0000000000000000 x16: ffffffde26f67cd0 x15: 0000000000000000
[   10.367028] x14: 0000000000000000 x13: 006f74634f20726f x12: 7463656a6e496f69
[   10.367036] x11: 0000000000000000 x10: 0000005cccae0cb8 x9 : ffffffddda656a24
[   10.367044] x8 : ffffff81082ec000 x7 : 0000000000000000 x6 : 000000000000003f
[   10.367052] x5 : 0000000000000000 x4 : ffffff81003c9ec0 x3 : ffffffddda665088
[   10.367060] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff81003c9ec0
[   10.367068] Call trace:
[   10.367070]  snd_card_init+0x34c/0x378 [snd]
[   10.367095]  snd_card_new+0x78/0xd0 [snd]
[   10.367119]  snd_soc_bind_card+0x360/0xbd8 [snd_soc_core]
[   10.367185]  snd_soc_register_card+0xf8/0x110 [snd_soc_core]
[   10.367226]  devm_snd_soc_register_card+0x54/0xb0 [snd_soc_core]
[   10.367266]  audioinjector_octo_probe+0x1d8/0x238 [snd_soc_audioinjector_octo_soundcard]
[   10.367277]  platform_probe+0x70/0xe0
[   10.367285]  really_probe+0xc4/0x2b0
[   10.367290]  __driver_probe_device+0x80/0xe8
[   10.367294]  driver_probe_device+0x44/0x110
[   10.367299]  __device_attach_driver+0xc0/0x100
[   10.367304]  bus_for_each_drv+0x84/0xd8
[   10.367308]  __device_attach+0xf4/0x188
[   10.367312]  device_initial_probe+0x1c/0x28
[   10.367317]  bus_probe_device+0xa4/0xb0
[   10.367321]  deferred_probe_work_func+0x90/0xc8
[   10.367325]  process_one_work+0x208/0x480
[   10.367333]  worker_thread+0x50/0x428
[   10.367338]  kthread+0xfc/0x110
[   10.367344]  ret_from_fork+0x10/0x20
[   10.367350] ---[ end trace 0000000000000000 ]---
[   10.368944] audioinjector-octo soc:sound: ASoC: driver name too long 'audioinjector-octo-soundcard' -> 'audioinjector-o'
[   12.293065] cs42xx8 1-0048: supply VA not found, using dummy regulator
[   12.293212] cs42xx8 1-0048: supply VD not found, using dummy regulator
[   12.293260] cs42xx8 1-0048: supply VLS not found, using dummy regulator
[   12.293310] cs42xx8 1-0048: supply VLC not found, using dummy regulator
[   12.307051] cs42xx8 1-0048: found device, revision 4
[   12.414781] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   12.414828] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.886768] ------------[ cut here ]------------
[   13.886784] WARNING: CPU: 1 PID: 764 at sound/core/init.c:309 snd_card_init+0x34c/0x378 [snd]
[   13.886832] Modules linked in: snd_soc_audioinjector_octo_soundcard(+) snd_soc_cs42xx8_i2c regmap_i2c snd_soc_cs42xx8 snd_seq_dummy snd_hrtimer snd_seq snd_seq_device overlay spidev rpivid_hevc(C) uinput bcm2835_codec(C) bcm2835_isp(C) brcmfmac bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops brcmutil snd_soc_bcm2835_i2s videobuf2_v4l2 cfg80211 videobuf2_common snd_soc_core videodev raspberrypi_hwmon snd_compress snd_pcm_dmaengine rfkill s [... cut ...]
[   14.641230] Bluetooth: HCI UART protocol H4 registered
[   14.641309] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.641505] Bluetooth: HCI UART protocol Broadcom registered
[   14.846026] Initializing XFRM netlink socket
[   14.885319] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.885340] Bluetooth: BNEP filters: protocol multicast
[   14.885358] Bluetooth: BNEP socket layer initialized
[   14.892025] Bluetooth: MGMT ver 1.22
[   14.902902] NET: Registered PF_ALG protocol family