Audio-Injector/Octo #68: rpi 5 - setup problem

clubwalk, 2024-10-14:
hello, i try to setup the octo for a week now and i can't make it works.
i'm on a rpi 5 and using the 3A power supply.

`iam@raspberrypi:~$ uname -a
Linux raspberrypi 6.6.56-v8-16k+ #1804 SMP PREEMPT Thu Oct 10 18:24:46 BST 2024 aarch64 GNU/Linux
liam@raspberrypi:~$ sudo dpkg -i Downloads/audioinjector.octo.setup_0.4_all.deb
(Reading database ... 151969 files and directories currently installed.)
Preparing to unpack .../audioinjector.octo.setup_0.4_all.deb ...
Unpacking audioinjector.octo.setup (0.4) over (0.4) ...
Setting up audioinjector.octo.setup (0.4) ...
updating the kernel
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:84e180662bc3f6646860a328d328a38d58325c8a
BOOTLOADER_REV:3c822369bec71a2f10514994b6f491d8e2323f9e
 *** Your firmware is already up to date (delete /boot/firmware/.firmware_revision and /boot/firmware/.bootloader_revision to force an update anyway)
setting up /boot/config.txt
backing up /boot/config.txt to /boot/config.txt.2024-10-14.13.18.13
setting up /boot/config.txt
backing up /etc/asound.conf to /etc/asound.conf.2024-10-14.13.18.13
backing up /root/.asoundrc to /root/.asoundrc.2024-10-14.13.18.13
backing up /home/pi/.asoundrc to /home/pi/.asoundrc.2024-10-14.13.18.13
/var/lib/dpkg/info/audioinjector.octo.setup.postinst: 85: cannot create ~pi/.asoundrc: Directory nonexistent
dpkg: error processing package audioinjector.octo.setup (--install):
 installed audioinjector.octo.setup package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 audioinjector.octo.setup
liam@raspberrypi:~$ 
`
thx by advance

clubwalk, 2024-10-14:
update : 
after erasing and imaging my rpi's sd, the setup was done without errors but the octo still not detected.

clubwalk, 2024-10-16:
went back to my 3b+, kernel 5.15.65-v8+, got all outputs working without noise.
hope this lil bud will handle my project.

snjee21, 2025-03-17:
1. OS: Raspberry PI OS Lite 64-bit
2. Install package
```
$ sudo dpkg -i ./audioinjector.octo.setup_0.4_all.deb
Selecting previously unselected package audioinjector.octo.setup.
(Reading database ... 83439 files and directories currently installed.)
Preparing to unpack .../audioinjector.octo.setup_0.4_all.deb ...
Unpacking audioinjector.octo.setup (0.4) ...
Setting up audioinjector.octo.setup (0.4) ...
updating the kernel
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:3608b77cd4557513506dbc098db04XXXXXXXXXXX
BOOTLOADER_REV:78d08e9763079c6608506e25eefeee5XXXXXXXXX
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 6.6.74+rpt-rpi-2712
WANT_32BIT:0 WANT_64BIT:1 WANT_PI4:1 WANT_PI5:1
##############################################################
WARNING: This update bumps to rpi-6.12.y linux tree
See discussions at:
https://forums.raspberrypi.com/viewtopic.php?t=379745
##############################################################
Would you like to proceed? (y/N)
Downloading bootloader tools
Downloading bootloader images
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  151M  100  151M    0     0  1062k      0  0:02:26  0:02:26 --:--:-- 1029k
BOOTLOADER: up to date
   CURRENT: Mon 10 Mar 17:10:37 UTC 2025 (1741626637)
    LATEST: Mon 10 Mar 17:10:37 UTC 2025 (1741626637)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 6.12.18-v8+
 *** depmod 6.12.18-v8-16k+
 *** Updating VideoCore libraries
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 3608b77cd4557513506dbc098db04938439804aa
 *** A reboot is needed to activate the new firmware
setting up /boot/config.txt
backing up /boot/config.txt to /boot/config.txt.2025-03-16.16.09.44
setting up /boot/config.txt
backing up /etc/asound.conf to /etc/asound.conf.2025-03-16.16.09.44
backing up /root/.asoundrc to /root/.asoundrc.2025-03-16.16.09.44
backing up /home/pi/.asoundrc to /home/pi/.asoundrc.2025-03-16.16.09.44
```
3. Insert "dtoveray" in config.txt
```
$ sudo nano /boot/firmware/config.txt
#dtparam=audio=on
dtoverlay=vc4-kms-v3d,noaudio
dtoverlay=audioinjector-addons,non_stop_clocks
```
4. reboot system
5. verify detect audio
```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: audioinjectoroc [audioinjector-octo-soundcard], device 0: AudioInject-HIFI cs42448-0 [AudioInject-HIFI cs42448-0]
 Subdevices: 1/1
 Subdevice #0: subdevice #0

$$dmesg | grep audio
[ 6.089483] vc4_hdmi 107c701400.hdmi: 'dmas' DT property is missing or empty, no HDMI audio
[ 6.097681] vc4_hdmi 107c706400.hdmi: 'dmas' DT property is missing or empty, no HDMI audio
[ 7.553847] audioinjector-octo soc@107c000000:sound: ASoC: driver name too long 'audioinjector-octo-soundcard' -> 'audioinjector-o'

$ dmesg | grep cs4*
[ 5.908821] cs42xx8 1-0048: supply VA not found, using dummy regulator
[ 5.908849] cs42xx8 1-0048: supply VD not found, using dummy regulator
[ 5.908865] cs42xx8 1-0048: supply VLS not found, using dummy regulator
[ 5.908873] cs42xx8 1-0048: supply VLC not found, using dummy regulator
[ 5.918203] cs42xx8 1-0048: found device, revision 4
```
[Conclusion]
In my case, when I play multiple wav and mp3 files, only similar waveforms are output.

[image]
https://github.com/user-attachments/assets/94e33288-8327-470b-a55c-2c7131941790

snjee21, 2025-03-17:
> hello, i try to setup the octo for a week now and i can't make it works. i'm on a rpi 5 and using the 3A power supply.
>
> `iam@raspberrypi:~$ uname -a Linux raspberrypi 6.6.56-v8-16k+ #1804 SMP [...]

If I check the error in your installation, in your case there is no "pi" user on rpi5. Create a pi user with the "adduser" command. Then the installation will pass normally.

clubwalk, 2025-04-04:
Thank you so much, it's finally installed! One problem stills remain, my audio output are totally unusable, making a periodic buzzing sound either within PureData or Web Browser

Pete1162, 2025-08-04:
> Thank you so much, it's finally installed! One problem stills remain, my audio output are totally unusable, making a periodic buzzing sound either within PureData or Web Browser

You need to downgrade the kernel. From 6 onwards it won't work.

See my comprehensive installation guide: https://github.com/Audio-Injector/Octo/issues/71

TommyKaneko, 2025-09-07:
I've finally got it working on my Raspberry Pi 3A and Zero2.  Here is a script that I adapted from Pete 1162 (see https://github.com/Audio-Injector/Octo/issues/71 ) to get it set up, with simple ALSA set up (No JackD or CamillaDSP - for that, go to https://github.com/Audio-Injector/Octo/issues/71 ).

The key step in the process is downgrading the Linux kernel to v5.15.34.  This, at the time of writing, seems to be the last kernel that worked with the Audio Injector Octo card.

### Raspberry Pi 3A / Zero2 with Audio Injector Octo – Installation Guide

This script installs the Octo audio interface on an RPi 3A or Zero2 without random channel swapping issues. All other audio interfaces of the Pi are disabled.

**Preparation:**
First, use the Pi Imager to select Raspberry Pi OS Lite 64-bit under "Raspberry Pi OS (other)." Then, under "Edit Settings," define your network access and username. Under "Services," enable SSH (password  #authentication).
Save and flash the SD card with these settings.
[image]

Now, save the following script as setup_octo_audio.sh using a text editor:

```
#!/bin/bash
#
# This script automates the installation of audio hardware and software
# for Octo audio on a Raspberry Pi running 64-bit Raspberry Pi OS Lite.
#
# It is designed to be idempotent, meaning it can be safely run multiple times.
# After a reboot is required, simply re-run the script to continue the setup.

# This script is adapted from a previous version to exclude CamillaDSP installation.

set -e

# Determine the effective user. Prioritize the first script argument, 
# then SUDO_USER, then the current USER. This makes the script's
# user context robust across re-executions.
if [ -n "$1" ]; then
    EFFECTIVE_USER=$1
else
    EFFECTIVE_USER=${SUDO_USER:-$USER}
fi
export EFFECTIVE_USER

STATE_FILE="/home/$EFFECTIVE_USER/.audio_setup_state"

# Function to log messages
log() {
    echo "[INFO] $(date +'%T') - $1"
}

# Function to check if a step is done
is_done() {
    grep -q "$1" "$STATE_FILE" 2>/dev/null
}

# Function to mark a step as done
mark_done() {
    echo "$1" >> "$STATE_FILE"
}

# --- Step Functions ---

step_1_update_system() {
    log "Step 1: Updating the system..."
    sudo apt update && sudo apt full-upgrade -y
    mark_done "step_1_update_system"
    log "System update complete."
}

step_2_configure_audio() {
    log "Step 2: Configuring audio in /boot/firmware/config.txt..."
    CONFIG_FILE="/boot/firmware/config.txt"

    # Ensure required dtparam entries exist and are not commented out
    sudo sed -i 's/^#*\s*dtparam=i2c_arm=on/dtparam=i2c_arm=on/' "$CONFIG_FILE"
[... 88 lines cut ...]
Open a terminal or use Windows PowerShell to connect to the Raspberry Pi via SSH.

```
ssh USERNAME@PI-IP
```

After entering the password, you'll be connected and automatically land in the user directory.
Now, make the script executable (no confirmation will be shown):

```
sudo chmod +x setup_octo_audio.sh
```
Now, run the script. You will occasionally be asked for confirmation. Always answer "Yes"! Perform restarts as required.
After a restart, run the script again—it will automatically continue from the correct point.
```
sudo ./setup_octo_audio.sh
```
After the final restart, check if ALSA correctly finds the interface with some of these commands:
```
aplay -l
arecord -l
aplay -D hw:0 --dump-hw-params
arecord -D hw:0 --dump-hw-params
alsamixer
```