Audio-Injector/Octo #71: Audio Injector Octo Installation Guide (with CamillaDSP and the JACK Audio API)

Pete1162, 2025-08-04:
Raspberry Pi 4 with Audio Injector Octo – Installation Guide

NOTE: This guide shows how to install the Octo audio interface along with CamillaDSP and the JACK Audio API on an RPi 4 without random channel swapping issues. All other audio interfaces of the Pi are disabled in this example. The result is a very low latency audio system.

1. Basic Installation
1.1 Install the Operating System:

    Use Raspberry Pi Imager

    OS: Raspberry Pi OS Lite (64-bit)

    Set up network, SSH, and user account (this guide assumes the username is pi; if using a different username, adjust commands accordingly).

1.2 Update the System:

sudo apt update && sudo apt full-upgrade -y

2. Audio Configuration
2.1 Edit config.txt:

sudo nano /boot/firmware/config.txt

#Edit these entries or add if non-existent:

dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
#dtparam=audio=on
dtoverlay=vc4-kms-v3d,noaudio

#at the very bottom of the file add:
dtoverlay=audioinjector-addons

#save and reboot

3. Install JACK Audio Server
3.1 Install JACK with real time permission:

sudo apt install jackd2

3.2 Configure Real-Time Priority:

sudo adduser $USER audio	#most likely already added

sudo nano /etc/security/limits.d/audio.conf

#Content looks like:

@audio   -  rtprio     95
@audio   -  memlock    unlimited
#@audio   -  nice      -19

3.3 Install Development Packages:

sudo apt install -y build-essential git cmake pkg-config libasound2-dev libjack-jackd2-dev libsamplerate0-dev

3.4 Set Up JACK Autostart Service:

sudo nano /etc/systemd/system/jackd.service

#Content:

[Unit]
Description=JACK Audio Daemon
After=alsa-restore.service
Requires=alsa-restore.service

[Service]
User=pi
Environment=JACK_NO_AUDIO_RESERVATION=1
LimitRTPRIO=95
LimitMEMLOCK=infinity
ExecStart=/usr/bin/jackd -P95 -d alsa -d hw:0 -r 48000 -p 128 -n 3
Restart=on-failure

[Install]
WantedBy=multi-user.target

#Enable and start the service:

sudo systemctl enable jackd.service
sudo systemctl start jackd.service
sudo reboot

#Check if Jack is running properly:

sudo systemctl status jackd.service

4. Install CamillaDSP
4.1 Install Rust and Cargo:

curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
#reboot

4.2 Install Dependencies:

sudo apt update
sudo apt install git

4.3 Clone CamillaDSP:

git clone https://github.com/HEnquist/camilladsp.git
cd camilladsp

4.4 Compile with JACK Support:

RUSTFLAGS='-C target-feature=+neon -C target-cpu=native' cargo build --release --features jack-backend,websocket

4.5 Create Service Files:

mkdir -p ~/camilladsp/coeffs ~/camilladsp/configs
sudo wget https://raw.githubusercontent.com/mdsimon2/RPi-CamillaDSP/main/camilladsp.service -O /lib/systemd/system/camilladsp.service

4.6 Adjust ExecStart in Service File and change username:

sudo nano /lib/systemd/system/camilladsp.service

#add your username username

#Change this line to:
ExecStart=/home/pi/camilladsp/target/release/camilladsp -s /home/pi/camilladsp/statefile.yml -w -g-0 -o /home/pi/camilladsp/camilladsp.log -p 1234

4.7 Enable and Start Service:

sudo systemctl enable camilladsp
sudo service camilladsp start

5. Install CamillaDSP GUI
5.1 Install Web Interface:

wget https://github.com/HEnquist/camillagui-backend/releases/download/v3.0.2/bundle_linux_aarch64.tar.gz -O ~/camilladsp/bundle_linux_aarch64.tar.gz
sudo tar -xvf ~/camilladsp/bundle_linux_aarch64.tar.gz -C /opt/

5.2 Set Up GUI Service:

sudo wget https://raw.githubusercontent.com/mdsimon2/RPi-CamillaDSP/main/camillagui.service -O /lib/systemd/system/camillagui.service
sudo nano /lib/systemd/system/camillagui.service  # Adjust username if needed

5.3 Enable and Start GUI Service:

sudo systemctl enable camillagui
sudo service camillagui start

5.4 Reboot:

sudo reboot

6. Verification

#Check status:

sudo systemctl status jackd.service
sudo systemctl status camilladsp
sudo systemctl status camillagui

#Web Interface:

http://<yourPI-IP>:5005/gui/index.html

7. Kernel Downgrade (Required, otherwise Octo will not work!)

sudo rpi-update f5c4fc199c8d8423cb427e509563737d1ac21f3c
sudo reboot

8. Camilla setup 	

	In the Camilla webgui under "Devices" > Capture & Playback Devices choose JACK
	Chunksize can be reduced for very low latency values - 32 is absolutely safe
	Create a new config under "Files" and mark it as active (the star button)
	On the left hand side click "Apply and save"
	You might need to adjust recording levels in alsamixer

DONE!

Pete1162, 2025-08-10:
`#!/bin/bash
#
# This script automates the installation of audio hardware and software
# for Octo audio on a Raspberry Pi running 64-bit Raspberry Pi OS Lite.
#
# It is designed to be idempotent, meaning it can be safely run multiple times.
# After a reboot is required, simply re-run the script to continue the setup.

set -e

STATE_FILE="$HOME/.audio_setup_state"

# Function to log messages
log() {
    echo "[INFO] $(date +'%T') - $1"
}

# Function to check if a step is done
is_done() {
    grep -q "$1" "$STATE_FILE" 2>/dev/null
}

# Function to mark a step as done
mark_done() {
    echo "$1" >> "$STATE_FILE"
}

# --- Step Functions ---

step_1_update_system() {
    log "Step 1: Updating the system..."
    sudo apt update && sudo apt full-upgrade -y
    mark_done "step_1_update_system"
    log "System update complete."
}

step_2_configure_audio() {
    log "Step 2: Configuring audio in /boot/firmware/config.txt..."
    CONFIG_FILE="/boot/firmware/config.txt"

    # Ensure required dtparam entries exist and are not commented out
    sudo sed -i 's/^#*\s*dtparam=i2c_arm=on/dtparam=i2c_arm=on/' "$CONFIG_FILE"
    grep -q "^dtparam=i2c_arm=on" "$CONFIG_FILE" || echo "dtparam=i2c_arm=on" | sudo tee -a "$CONFIG_FILE" > /dev/null

    sudo sed -i 's/^#*\s*dtparam=i2s=on/dtparam=i2s=on/' "$CONFIG_FILE"
    grep -q "^dtparam=i2s=on" "$CONFIG_FILE" || echo "dtparam=i2s=on" | sudo tee -a "$CONFIG_FILE" > /dev/null

    sudo sed -i 's/^#*\s*dtparam=spi=on/dtparam=spi=on/' "$CONFIG_FILE"
    grep -q "^dtparam=spi=on" "$CONFIG_FILE" || echo "dtparam=spi=on" | sudo tee -a "$CONFIG_FILE" > /dev/null

    # Ensure dtparam=audio=on is commented out
    sudo sed -i 's/^\s*dtparam=audio=on/#dtparam=audio=on/' "$CONFIG_FILE"

    # Ensure vc4-kms-v3d,noaudio overlay is present and not duplicated
    if grep -q "^dtoverlay=vc4-kms-v3d" "$CONFIG_FILE"; then
        # Line exists, ensure it's correct
        sudo sed -i 's/\(dtoverlay=vc4-kms-v3d\).*/\1,noaudio/' "$CONFIG_FILE"
    else
        # Line does not exist, add it
        echo "dtoverlay=vc4-kms-v3d,noaudio" | sudo tee -a "$CONFIG_FILE" > /dev/null
    fi

    # Ensure audioinjector-addons overlay is present at the end
    grep -q "^dtoverlay=audioinjector-addons" "$CONFIG_FILE" || echo -e "\n# For Octo Audio\ndtoverlay=audioinjector-addons" | sudo tee -a "$CONFIG_FILE" > /dev/null

    mark_done "step_2_configure_audio"
    log "Audio configuration complete. A reboot is required."
    read -p "Press Enter to reboot now, or Ctrl+C to reboot later." -n 1 -s
    sudo reboot
}
[... 163 lines cut ...]
    step_3_install_jack
fi

if ! is_done "step_4_install_camilladsp"; then
    step_4_install_camilladsp
fi

if ! is_done "step_5_install_camillagui"; then
    step_5_install_camillagui
fi

if ! is_done "step_6_kernel_downgrade"; then
    step_6_kernel_downgrade
fi

log "All setup steps are complete!"
log "You can verify the services with:
    sudo systemctl status jackd.service
    sudo systemctl status camilladsp
    sudo systemctl status camillagui"
log "The web interface should be available at: http://<your-pi-ip>:5005/gui/index.html"

rm "$STATE_FILE"
exit 0
`

Pete1162, 2025-08-10:
I've created a script for easy installation. See this post here (use translator if needed)

https://www.diy-hifi-forum.eu/forum/showthread.php?24280-Kleiner-Guide-CamillaDSP-plus-Audio-Injector-Octo-6x8-Interface-an-Raspberry-Pi-4&p=356691&viewfull=1#post356691

TommyKaneko, 2025-09-02:
> Voilà—you’ve installed the Octo interface with CamillaDSP, JACK, and a simple 2x2 setup!

Firstly, thank you for contributing such a comprehensive script to get the Octo going.

I will be trying this on my RPi3A and get back to you with results.  My one question: when you say a "simple 2x2 setup", that was only your test yes?  I will be able to set up all 8 channels (out) or 6 channels (in), yes?

TommyKaneko, 2025-09-07:
Thank you Pete for this.  I was on a Raspberry Pi 3A/Zero2, and I could not get Camilla DSP to compile:
```
Compiling CamillaDSP v3.0.1 (/home/tomkaneko/camilladsp)
error: could not compile `CamillaDSP` (lib)

Caused by:
  process didn't exit successfully: ....
```
I also needed to run the compilation several times to get various packages compiled with Cargo.  Can I suggest you add `--keep-going` flag to your line:
```
    RUSTFLAGS='-C target-feature=+neon -C target-cpu=native' cargo build --release --keep-going --features jack-backend,websocket
```
...but I simply could not get the compilation of CamillaDSP to work on the 3A or Zero2.  I think it is because of RAM limitations of the earlier models that times out the compilation process.  I expect if I setup the SD card on a Pi4 using your script, then move the card to 3A, there would be no problems, but I don't have a Pi4.

I ended up excluding JackD and CamillaDSP (skipping your steps 3 to 5) to get a simple ALSA setup.  This suited me fine for my purposes.  I have shared an adapted version of your script on https://github.com/Audio-Injector/Octo/issues/68 .

Incidentally, I got a question from JackD installation about whether it could take 'realtime priority'.  Saying yes I think stopped me from controlling ALSA directly with `arecord`, so that is why I didn't install JackD either.

But thank you for getting me on the path to get the AudioInjector Octo working!  It has taken me a full week!

Pete1162, 2025-09-07:
Hi Tommy,

I've ordered another Octo alongside a 3A+. I will tinker a bit and see if I can get it to run as intended in the first place.

I don’t know if the channel swapping issues that other users here are reporting occur without JackD. I was only able to get it under control with Jack. Since I'm using Camilla as a crossover it's absolutely crucial to have a persistent channel routing.